ctu-mrs/mrs_uav_system #9: ROS Noetic compatibility checklist

klaxalk, 2020-06-25:
We have started the ROS Noetic upgrade. Here is the checklist of stuff that needs to happen:

Automated install scripts:
- [x] uav_core install script
- [x] simulation install script
- [x] mrs_uav_system install script
- [x] linux_setup install script

Compilation compatibility:
- [x] uav_core compiles
- [ ] simulation compiles
- [ ] mrs_uav_system compiles and runs

Simulation compatibility:
- [x] mrs_gazebo_common_resources compiles
- [ ] px4/sitl_gazebo compiles

klaxalk, 2020-06-25:
We are currently stuck with building te px4/sitl_gazebo. The problem is with the Mavlink protocol. We have in total 3 programs which rely on Mavlink:
- Mavros
- px4/firmware
- px4/sitl_gazebo
We compile Mavlink ourselves and we can choose whatever version suits us.
However, we weren't able to find a single version that would currently work with all three simultaneously, given the px4/sitl_gazebo is updated to the latest version (as of today, 2020/06/25).
Sadly, the authors of px4/sitl_gazebo do not state for which version is their current _master_ prepared.
But since it looks like they are actively working on it, let's wait for them to update to Gazebo 11.

dookei, 2020-10-22:
Hi @klaxalk I have used v1.11.0 release with Gazebo 11 without any issue.

spurnvoj, 2020-10-22:
@klaxalk As the backup solution for people under Ubuntu 20.04, mavros can be installed using apt. That seems to work :)

klaxalk, 2020-10-24:
@dookei it is ok now, we are able to compile everything. However, now there is a problem with running things. We struggle with **ros::Timers**, which refuse to run of their rate is over 10 Hz. Will need some debugging.

closed 2020-12-21 by klaxalk