ctu-mrs/mrs_uav_system #11: Installation

AiltonOliveir, 2020-08-17:
i was following the installation tutorial, but the topic:

"- link our packages to the workspaces"

it's not clear for me, I'm new using ROS, if you can detail this step, I would be grateful.

Best regards

petrapa6, 2020-08-18:
Hello,

I suppose you are following the [I have a fresh Ubuntu and want it quick and easy](https://github.com/ctu-mrs/mrs_uav_system#i-have-a-fresh-ubuntu-1804-and-want-it-quick-and-easy) installation. This process is automated such that it will do all the listed parts without a need for your intervention. To run the automatic installation, just copy+paste the code into your terminal and press enter.

In particular:
- Linking our packages to the workspaces means that [symbolic links](https://kb.iu.edu/d/abbe) of **$GIT_PATH/uav_core**, **$GIT_PATH/simulation** will be created within a workspace (i.e., **~/mrs_workspace/src**). If installed, **$GIT_PATH/example_ros_packages** will also be linked to **~/workspace/src**.
- The workspaces are build using [catkin tools](http://wiki.ros.org/catkin) -> see our [How to compile](https://ctu-mrs.github.io/docs/introduction/how_to_compile.html) manual.
- **~/.bashrc** is a shell script, which is run on every interactive shell launch such as opening a terminal. See [this](https://unix.stackexchange.com/questions/129143/what-is-the-purpose-of-bashrc-and-how-does-it-work) for more information. This script is particularly important as we use it to [source ROS and workspaces, and to define global variables](https://github.com/ctu-mrs/uav_core#finishing-your-bashrc). More information on _sourcing_ is available [here](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment).

If you are new to ROS, I recommend taking at least a look at the beginner [ROS tutorials](http://wiki.ros.org/ROS/Tutorials) and reading our [wiki](https://ctu-mrs.github.io/).

AiltonOliveir, 2020-08-18:
Thank you for clarifying.

closed 2020-08-18 by AiltonOliveir