ctu-mrs/mrs_uav_system #15: component list F450 quadrotor

bconvens, 2020-09-28:
Dear MRS team,

Would it be possible to provide the component lists that I would need to order to build and fly the exact same F450 drone as you have?
- 450 frame kit
- motors
- props
- ESCs
- battery
- radio,    do I need a radio for each member of the swarm?
- receiver
- companion computer: which exact intel NUC specs (CPU, RAM, ...) would be the least that has enough compute power for running your software? Which one do you use.
- pixhawk 4
- gps? which rtk gps do you use?
- special cabling required?
- other sensors?
- anything else?

Is there anything that you custom 3D printed? Would you mind sharing the .STL files for 3D printing?

Thanks!
Best,
Bryan

DanHert, 2020-10-23:
Hello, sorry for the late reply.

- F450 frame kit, we buy the copies from china since the original is not made anymore
- Motors are correct
- We use the original DJI props, we found out that some of the copies have worse performance, but they still work
- We use these ESCs: https://hobbyking.com/en_us/turnigy-multistar-32bit-51a-race-spec-esc-2-6s-opto.html?queryID=b8f5dd1784099a8f3e6aa86811b2b4a1&objectID=74316&indexName=hbk_live_magento_en_us_products  There is no problems with them.
- Battery is Tattu 4S 6750 mAh 25C
- Receiver is Optima SL, transimtter is Hitec Flash 8, each drone has to have its own receiver, but with larger swarms we have just one transmitter connected to all the receivers.
-  Pixhawk 4 plastic case is fine, all our drones have them.
- Any SD card will do for pixhawk, it is just for config files and low rate logging.
- Currently we recommend NUC10i7FNK with 16 GB RAM (8 GB works too, but compilation can sometimes run out of memory, but no issues during runtime) and at least 128 GB SSD, more if you want to record a lot of data from your flights. We buy the NUCs without RAM and SSD, then disassemble them, install the RAM and SSD and mount them on the drones. The NUC is powered straight from the battery, it has 12-19V input voltage range, so a 4S battery can be plugged right in the NUC. AMD can also run our system. The system is using multiple cores, so more cores is good.
- We are using the regular FTDI clone boards like this - https://www.aliexpress.com/item/4001323460121.html?spm=a2g0o.productlist.0.0.7b83e8dcbCXILT&algo_pvid=1c2070da-7f8f-47ad-a8a0-83f3b4ca9cb6&algo_expid=1c2070da-7f8f-47ad-a8a0-83f3b4ca9cb6-5&btsid=2100bdd016034491508824193e00b8&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
One end is connected to the TELEM2 port, the other goes to the NUC USB. The NUC has two USB ports on Picoblade headers, we use those to connect the FTDI board (this is a much tougher connection, but it requires to make a custom cable, desolder the mini USB from the FTDI board and replace it with directly soldered cables).
- NUC already has a WIFI module integrated, we use that one.
- For bigger swarm experiments, we attach a WIFI router to one of the drones, so that they can carry it with them, otherwise we leave it somewhere on the ground. We use this WIFI router - TP-LINK TL-WR902AC as it can be powered through USB (directly from the NUC or from a powerbank or laptop).  
- We use this GPS - https://shop.holybro.com/pixhawk-4-gps-module_p1094.html
- We have the Emlid Reach 2 RTK on order right now. We used the first generation and it works well, so I can recommend it.
- Other sensors include Garmin Lidar Lite to measure altitude, Bluefox  global shutter cameras for UVDAR/OpticFlow/VIO/Other stuff, Realsenses, RPLidar A3, Ouster 3D lidars, Terabee and FLIR thermal camers and more, I cannot recall all of them :D.
- We do not use any telemetry modules, we communicate with the drone over WIFI, or with the RC transmitter.
- We do not have any dedicated ground stations, only thing you need is just a laptop and a WIFI router.
- We print most of the sensor mounts, the landing legs, battery holders and other things. I made a new repo for the 3D models - https://github.com/ctu-mrs/3D_models and put some of the F450 models there. If you need other models we can add them as well.

bconvens, 2020-10-26:
Thanks a lot @DanHert for the detailed responses!

Some follow up questions:
- should your ESCs be used in order to set  [the required ESC configs](https://github.com/ctu-mrs/uav_core/blob/master/miscellaneous/blheli32_esc_config/T650/T650_M1.ini)? Would [these cheaper ESCs](https://www.banggood.com/Racerstar-RS30A-Lite-30A-Blheli_S-BB1-2-4S-Brushless-ESC-Support-Dshot150-Dshot300-for-RC-FPV-Racing-Drone-p-1100236.html?rmmds=myorder&cur_warehouse=CN) be giving the same results or does it require some change of the config parameters? 
- do you think [this cheaper transmitter](https://www.banggood.com/Flysky-FS-i6X-i6X-10CH-2_4GHz-AFHDS-2A-RC-Transmitter-With-FS-iA10B-Receiver-for-FPV-RC-Drone-p-1101154.html?rmmds=myorder&ID=42481&cur_warehouse=CN) is ok. Or does your transimtter Hitec Flash 8 offer some additional features which are required?
- For what concerns the pixhawk's SD card I read [here](https://dev.px4.io/v1.9.0/en/log/logging.html) dropouts can result in wrong logged data. PX4 proposes the SanDisk Extreme U3 32GB, since it is least sensitive to dropouts. Did you experience dropouts and do you have an effective strategy to get rid of them?
- I had already ordered [these boards](https://www.amazon.com/gp/product/B00LODGRV8/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), do you think they should work too?
- So with 1 x $1899 [REACH RS2](https://emlid.com/reachrs2/) the pixhawk 4 GPS on each drone in a swarm becomes centimeter precise? Or do you need >1 base station? What's the reason you upgraded from the [REACH RS+](https://emlid.com/reachrs/)? 

Thanks!
Bryan

DanHert, 2020-10-26:
- The other ESCs should be fine, there is no need to use our exact config
- We only use Hitec Flash or Hitec Aurora transmitters which work well. I cannot guarantee you if the cheap one will work too, but probably yes.
- We only check the Pixhawk logs when we have a crash/failsafe land which cause we cannot deduce from the outputs of the mrs_uav_system. And if the card is not full, the logs are always there, never been a problem. We use the microSD card which comes with the Pixhawk 4.
- yes, these boards shoud be ok.
- You need one RS2 as a basestation, and then each drone needs its own reach M2 + LoRa radio for corrections. The Reach RS+ is a singleband receiver, while RS2 is a multiband one, which means faster fix time, longer possible baseline, better performance in worse conditions (blocked sky) etc...

closed 2020-11-18 by DanHert

bconvens, 2021-01-25:
Hi @DanHert ,

Which cabling and connectors do you use to power up and transmit signals from the Emlid reach M2 and the Lora radio to other components of the drone F450 drone?
Which Emlid specific cabling/connectors do you use (if any) and which cabling/connectors did you buy elsewhere for making your RTK solution work?

Thanks for the help!
Best,

DanHert, 2021-01-25:
Hello, I tried to power the Emlid directly from pixhawk, but pixhawk does not provide enough power. So our solution is to connect the emlid with USB to the NUC. So no additional cables are needed, just the one to connect lora and then a USB cable to the NUC.

DanHert, 2021-01-25:
Btw this also means that the RTK data will not get into pixhawk, only the NUC. So if you want to fly with RTK in feedback, you need to fly autonomously using the onboard computer, pixhawks position hold will still only use the regulart GPS. You can pipe the rtk data into pixhawk too, you just need an aditional cable.

bconvens, 2021-01-25:
@DanHert , I want to fly using RTK gps as position feedback using mrs uav system. So if I understood you correctly,  I just need 1 usb from emlid reach M2 to NUC and I don't need to pipe RTK data from NUC to pixhawk?

DanHert, 2021-01-25:
Yes, that is correct.