ctu-mrs/mrs_uav_system #17: Simulation termination

donkarlo, 2020-10-27:
I thought closing the terminal in which I run a simulation such as that in three_drones_gps will terminate the whole simulation environment but it seems it doesn't do so. That is when I close the terminal and Gazebo and I want to start another simulation gazebo  then I keep receiving errors and gazebo doesn't open up. May you please let me know what is the proper way of terminating one simulation and going to the other?

DanHert, 2020-10-27:
Hello, indeed, just closing the terminal will not kill the tmux session the simulation is running in.
To kill it properly, go to the terminal window and hit "CTRL+a" and then hit "k" (release "CTRL+a" before hiting "k").
This will kill the tmux session.

You can see all the running tmux sessions by typing "tmux ls" into you terminal, and then you can attach into those sessions
by typing "tmux a -t $SESSION_NAME". 

Cheers, Dan H.

klaxalk, 2020-10-27:
Hey, I recommend our cheatsheet: https://github.com/ctu-mrs/mrs_cheatsheet

donkarlo, 2020-10-28:
@DanHert  Thanks a lot. It worked very well. @klaxalk Thanks a lot

closed 2020-11-04 by klaxalk