ctu-mrs/mrs_uav_system #18: Can not fly higher then 30m (simulation)

maxhofidrone, 2020-11-03:
Hi,
I'm starting to use the MRS UAV System and could successfully use it in simulation. But I was unable to fly higher than 30m from ground.
I used following simulation:` mrs_workspace/src/simulation/example_tmux_scripts/one_drone_gps/start.sh`
and adopted the parameter `max_height: 50.0` (it use to be 15) in `/mrs_workspace/src/uav_core/ros_packages/mrs_uav_general/config/worlds/world_simulation.yaml` 
I now would expect to be able to fly to a max. height of 50m. But the UAV always stopped at 30 m. 

To fly higher as 30m, I needed to set the parameter `max_height_checking`  to false  in `/mrs_workspace/src/uav_core/ros_packages/mrs_uav_managers/config/default/uav_manager.yaml`

Do I miss some other configuration where the max. height is set to 30? 
I would appreciate any help!

donkarlo, 2020-11-03:
Hi I think you should check and change the constraints.I know checking the constraints could be done by **rostopic echo /uav1/control_manager/current_constraints** setting constraints must be easy but as of writing this, I have not yet checked.I will update you if I find something.

klaxalk, 2020-11-03:
@donkarlo definitely not, those are dynamic constraints. There is no maximum height/altitude in there (which you can check by `rostopic echo /uav1/control_manager/current_constraints`). Please, double-check your claims first before giving fast advice.

@maxhofidrone Hey, yes there is another way how the maximum height/altitude is limited. The state estimator node ([mrs_uav_odometry](https://github.com/ctu-mrs/mrs_uav_odometry)) can enforce maximum height based on the current estimator. In the end, the actual maxim height is the minimum from the one in the world file (you found that correctly) and the one specified by the odometry. Now I see this feature is not well-documented and definitely not well-built in general, I will put in on TODO.

The parameter can be changed in [default_config.yaml](https://github.com/ctu-mrs/mrs_uav_odometry/blob/master/config/default_config.yaml) and it is the:
```yaml
altitude:
  # Altitude limits
  max_default: 30.0 # TODO load from world file?
```

donkarlo, 2020-11-03:
@klaxalk hi i removed the post. just wanted to share what i know. but i also said probably so that the reader checks himself as well.

maxhofidrone, 2020-11-03:
@klaxalk thank you very much. That solved it!

klaxalk, 2020-11-03:
@donkarlo ye, just... why to advise somene to check a topic (which you somehow know) if you don't know if the information is in there. Check the topic first for yourself and then give the advice. And you don't need to run the system for this. For others as well: if you want to see the list of our common topics, check [https://ctu-mrs.github.io/docs/system/uav_ros_interface.html](https://ctu-mrs.github.io/docs/system/uav_ros_interface.html). This will bring you directly to the message type of [mrs_msgs/DynamicConstraints](https://ctu-mrs.github.io/mrs_msgs/msg/DynamicsConstraints.html).

donkarlo, 2020-11-03:
@klaxalk Just was excited to share a piece of information in which the data could be found. I will certainly not do it again even if I mention the word probable in advance.

petrlmat, 2020-11-03:
https://github.com/ctu-mrs/mrs_uav_odometry/commit/f09697580992b07d58baa66afd02d7fac98ee916 should simplify this. The maximum height above ground is now by default taken from the world file, unless the parameter in the world file does not exist.

closed 2020-11-04 by klaxalk