ctu-mrs/mrs_uav_system #20: UAV crashes after regaining valid rangefinder measurments + custom config for altitude estimation has no effect (Simulator)

maxhofidrone, 2020-11-06:
Hi,
I encounter following problem:

I use the trajectory loader to follow a simple descending parabola shaped trajectory  fro 35m to 5m (ref. to takeoff alt.). For that I'm using an adopted version of the one_drone_gps simulation with changed max_altitude and no safety area. 

When flying over 20m above ground the _control_ panel shows following warnings:
```
[ INFO] [1604676334.599842377, 633.972000000]: [Odometry]: garmin measurement not passed through gate: 22.303928                                                                                          
[ INFO] [1604676335.144893111, 634.516000000]: [ControlManager]: tracker: 'MpcTracker', controller: 'Se3Controller', mass: '3.59 kg', disturbances: body [-0.76, 0.34] N, world [-0.84, 0.36] N           
[ INFO] [1604676335.144993051, 634.516000000]: [Odometry]: Running for 624.74 s. Estimators: Lat: GPS, Alt: HEIGHT, Hdg: PIXHAWK. GPS Cov: 1.00. Max alt: 150.00.                                         
[ WARN] [1604676335.608980602, 634.980000000]: [Odometry]: garmin range value 21.666801 > 20.000000 is not valid.                                                                                         
[ INFO] [1604676335.609148178, 634.980000000]: [Odometry]: garmin measurement not passed through gate: 21.666801                                                                                          
[ WARN] [1604676336.616339746, 635.988000000]: [Odometry]: garmin range value 21.018160 > 20.000000 is not valid. 
```
As soon as the _garmin range value_ is below 20m follwing output is displayed:

```
[ WARN] [1604676338.285378738, 637.656000000]: [Odometry]: Garmin measurement 19.986118 declined by median filter.
[ERROR] [1604676338.478850455, 637.848000000]: [ControlManager]: activating failsafe land: control_error=7.22/6.00 m (x: -0.00, y: 0.00, z: -7.22)
[ INFO] [1604676338.478959482, 637.848000000]: [ControlManager]: calling for parachute deployment
[ERROR] [1604676338.479498368, 637.848000000]: [ControlManager]: service call for parachute deployment failed!
[ERROR] [1604676338.479569638, 637.848000000]: [ControlManager]: could not deploy parachute: 'error during deployment of parachute', continuing with normal failsafe
[ INFO] [1604676338.479623397, 637.848000000]: [ControlManager]: activating the controller 'FailsafeController'
[ INFO] [1604676338.479785965, 637.848000000]: [FailsafeController]: activated with yaw: 0.00 rad
[ INFO] [1604676338.479828549, 637.848000000]: [FailsafeController]: activated with uav_mass_difference 0.08 kg.
[ INFO] [1604676338.479916108, 637.848000000]: [ControlManager]: switching odometry callbacks to OFF
[ INFO] [1604676338.486727635, 637.856000000]: [Odometry]: Callbacks disabled
[ INFO] [1604676338.486898794, 637.856000000]: [ControlManager]: the controller 'FailsafeController' was activated
[ INFO] [1604676338.486957629, 637.856000000]: [Se3Controller]: deactivated
[ INFO] [1604676338.487049627, 637.856000000]: [FailsafeController]: first iteration
[ INFO] [1604676338.490578627, 637.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 3.23
[ INFO] [1604676339.489916445, 638.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 3.02
[ WARN] [1604676340.004416382, 639.372000000]: [Odometry]: Garmin measurement 17.308140 declined by median filter.
[ INFO] [1604676340.148742286, 639.516000000]: [Odometry]: Running for 629.74 s. Estimators: Lat: GPS
, Alt: HEIGHT, Hdg: PIXHAWK. GPS Cov: 1.00. Max alt: 150.00.[ INFO] [1604676340.148885639, 639.516000000]: [ControlManager]: tracker: 'MpcTracker', controller: 'FailsafeController', mass: '3.58 kg', disturbances: body [0.00, 0.00] N, world [0.00, 0.00] N
[ INFO] [1604676340.492546805, 639.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 2.81
[ WARN] [1604676341.012645975, 640.380000000]: [Odometry]: Garmin measurement 12.334932 declined by median filter.
[ INFO] [1604676341.492292949, 640.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 2.61
[ WARN] [1604676342.020744635, 641.388000000]: [Odometry]: Garmin measurement 2.918188 declined by median filter.
[ WARN] [1604676342.281253513, 641.648000000]: [ControlManager]: tilt error exceeded threshold (28.15/20.00 deg)
[ WARN] [1604676342.285460949, 641.652000000]: [Odometry]: Excessive tilt detected: 0.59 rad. Not fusing.
[ WARN] [1604676342.289750796, 641.656000000]: [ControlManager]: tilt error (33.52 deg) over threshold for 0.01 s
[ WARN] [1604676342.389527684, 641.756000000]: [ControlManager]: tilt error (67.80 deg) over threshold for 0.11 s
[ WARN] [1604676342.489883078, 641.856000000]: [ControlManager]: tilt error (51.85 deg) over threshold for 0.21 s
[ INFO] [1604676342.494363078, 641.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 2.42
[ WARN] [1604676342.589699819, 641.956000000]: [ControlManager]: tilt error (25.75 deg) over threshold for 0.31 s
[ WARN] [1604676343.036523215, 642.392000000]: [Odometry]: Garmin measurement 0.699748 declined by median filter.                                                                                         
[ WARN] [1604676343.296447759, 642.656000000]: [ControlManager]: tilt error exceeded threshold (23.79/20.00 deg)                                                                                          
[ WARN] [1604676343.300450890, 642.660000000]: [Odometry]: Excessive tilt detected: 0.53 rad. Not fusing.                                                                                                 
[ WARN] [1604676343.308458479, 642.668000000]: [ControlManager]: tilt error (30.25 deg) over threshold for 0.01 s                                                                                         
[ WARN] [1604676343.408383746, 642.768000000]: [ControlManager]: tilt error (68.23 deg) over threshold for 0.11 s                                                                                         
[ INFO] [1604676343.500360495, 642.860000000]: [ControlManager]: failsafe: initial mass: 3.58 thrust_mass_estimate: 2.23                                                                                  
[ WARN] [1604676343.508941482, 642.868000000]: [ControlManager]: tilt error (85.42 deg) over threshold for 0.21 s                                                                                         
[ WARN] [1604676343.620508514, 642.976000000]: [ControlManager]: tilt error (68.75 deg) over threshold for 0.32 s                                                                                         
[ WARN] [1604676343.719702639, 643.076000000]: [ControlManager]: tilt error (46.72 deg) over thresho$d for 0.42 s                                                                                         
[ERROR] [1604676343.808831664, 643.168000000]: [ControlManager]: tilt error too large for 0.51 s, disarming                                                                                               
[ INFO] [1604676343.808880800, 643.168000000]: [ControlManager]: switching motors OFF                
[ INFO] [1604676343.808902269, 643.168000000]: [ControlManager]: switching to 'NullTracker' after switching motors off                                                                                    
[ INFO] [1604676343.808931599, 643.168000000]: [ControlManager]: activating the tracker 'NullTracker'
[ WARN] [1604676343.808959122, 643.168000000]: [ControlManager]: motors are off   
```

The range measurement is rejected, resulting in a failsafe action which results into:

1. a recovery when simulating in the standard "grass" world
2. crash when simulating in a custom hilly environment, even if the same "space" under the drone is provided at this point. (Above's log is from the 2. case)

I am able the avoid this crash if I am changing the altitude estimator in the status panel to "BARO".
Does anyone has an idea why the state estimation breaks using the rangefinder measurements? Are there additional configs? They obviously work before flying above 20m AGL.

My (related) custom config problem:
I now try to set custom configs for the altitude estimator according this [HowTo](https://ctu-mrs.github.io/docs/system/custom_configs.html#how-to-use-custom-config-files-for-uav_core-modules). It works with the provided example. In the same manner I want now to change altimeter estimator after takeoff with a custom config file: ./custom_configs/config_odometry.yaml:

`takeoff:`
`  after_takeoff:`
`    altitude_estimator: "BARO"`

Unfortunately this does not have any effects. As asked in the [HowTo](https://ctu-mrs.github.io/docs/system/custom_configs.html#how-to-use-custom-config-files-for-uav_core-modules) i load the custom config file in the session.yaml:
`config_uav_manager:=./custom_configs/config_odometry.yaml`

I would appropriate any help and hints to understand this problem!

Best,

Max

petrlmat, 2020-11-06:
Hi,
this behavior is implemented due to the unreliability of the garmin sensor above 20 m height. When flying above 20 m, the UAV relies on barometer measurements only. Barometer measures the ASL altitude, so when the UAV returns below 20 m, the AGL measurements of rangefinder are inconsistent with the ASL state.

To fly in a hilly environment, it is necessary to use the BARO estimator instead of HEIGHT, because the rangefinder measures AGL height, which can be useful when you want to follow the terrain elevation, but can also cause problems when the terrain below the drone abruptly changes (e.g., flying above a cliff). In this situation, the controller would try to get to the desired AGL height by minimizing the control error. We detect this change abrupt change in height and instead of allowing the controller to perform a dangerous maneuver, a failsafe is performed. 

So your idea to change the estimator from HEIGHT to BARO is correct. The only problem is, that you are setting custom config for the `uav_manager` node, but you need to change it for the `odometry` node like this:
`config_odometry:=./custom_configs/odometry.yaml`

Another option is to add:
```
altitude:
  gate:
    garmin:
      max: 40.0
```
to your odometry custom config (careful with the indentation), since in simulation the rangefinder performs equally in its whole measurement range.

Hope this helps

Matej

maxhofidrone, 2020-11-10:
Dear @petrlmat 
thank you very much! I understand it now.

Best regards,

Max

petrlmat, 2020-11-10:
Glad to help.

Matej

closed 2020-11-10 by petrlmat