ctu-mrs/mrs_uav_system #23: Installation Issues

lucasmartins131, 2020-11-23:
I am very begining using ROS and I am trying to use your system for simulation some uavs
I installed using the script at the installation page and then tried to simulate via two ways: 

1) Using one the scripts and I get the error: 
export UAV_NAME=uav1; export RUN_TYPE=simulation; export UAV_TYPE=t650; export WORLD_NAME=simulation; export STANDALONE=true; export SENSORS="garmin_down"
waitForSimulation; roslaunch mrs_uav_status status.launch
lucas@lucas:~/mrs_workspace/src/simulation/example_tmux_scripts/one_drone_gps_standalone$ export UAV_NAME=uav1; export RUN_TYPE=simulation; export UAV_TYPE=t650; export WORLD_NAME=simulation; export STANDALONE=true; export SENSORS="garmin_down"
lucas@lucas:~/mrs_workspace/src/simulation/example_tmux_scripts/one_drone_gps_standalone$ waitForSimulation; roslaunch mrs_uav_status status.launch
waiting for simulation
waiting for simulation

2) Launching an mrs_simulation and trying to spwan a drone. And the I get the following error: 
lucas@lucas:~$ roslaunch mrs_simulation simulation.launch gui:=true
RLException: [simulation.launch] is neither a launch file in package [mrs_simulation] nor is [mrs_simulation] a launch file name
The traceback for the exception was written to the log file

Probably I got an issue during the installation procedure, but I don´t know how to correct that and starting using the mrs 

Thank you very much

DanHert, 2020-11-23:
Looks like there was some problem with the installation of the simulation. If you go to` ~/mrs_workspace/src` is there a symlink to the simulation? If not, you should add it (go to the `~/mrs_workspace/src` and call `ln -s ~/git/simulation`), then install the simulation again (just to be sure) - run `~/git/simulation/installation/install.sh`. Recompile the workspace (go to `~/mrs_workspace` and call `catkin clean` and then `catkin build`).

After this, try to run one of the example tmux scripts.

lucasmartins131, 2020-11-23:
Dan I decided befor reading your advices to reinstall all the package again, and then I got several compilation errors: 
I will post some: 
Errors     << mrs_gazebo_common_resources:make /home/lucas/mrs_workspace/logs/mrs_gazebo_common_resources/build.make.000.log                                                                                        
c++: internal compiler error: Morto (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/MRSGazeboCameraPlugin.dir/src/sensor_and_model_plugins/camera_plugin.cpp.o] Error 4
make[1]: *** [CMakeFiles/MRSGazeboCameraPlugin.dir/all] Error 2
make[1]: ** Esperando que outros processos terminem.
c++: internal compiler error: Morto (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/MRSGazeboRealsensePlugin.dir/src/sensor_and_model_plugins/realsense_plugin.cpp.o] Error 4
make[1]: *** [CMakeFiles/MRSGazeboRealsensePlugin.dir/all] Error 2
make: *** [all] Error 2
cd /home/lucas/mrs_workspace/build/mrs_gazebo_common_resources; catkin build --get-env mrs_gazebo_common_resources | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
....................................................................................................................................................................................................................
Failed     << mrs_gazebo_common_resources:make           [ Exited with code 2 ]                                                                                                                                     
Failed    <<< mrs_gazebo_common_resources                [ 5 minutes and 27.9 seconds ]                                                                                                                             
Abandoned <<< mrs_simulation                             [ Depends on failed job mrs_gazebo_common_resources ]                                                                                                      
Finished  <<< mavros_extras                              [ 5 minutes and 38.1 seconds ]                                                                                                                             
Starting  >>> mrs_lib                                                                                                                                                                                               
Starting  >>> test_mavros                                                                                                                                                                                           
_____________________________________________________________________________________________________________________

DanHert, 2020-11-23:
Hmmm, this looks like a lower level problem. What kind of computer do you use (CPU and how much RAM)? And what version of Ubuntu are you on?

lucasmartins131, 2020-11-23:
Memory
          descrição: Memória do sistema
          ID físico: 0
          tamanho: 5882MiB
     *-cpu
          produto: AMD Ryzen 7 3700U with Radeon Vega Mobile Gfx
          fabricante: Advanced Micro Devices [AMD]
            informações do barramento: cpu@0
          size: 3434MHz
          64 bits
Ubuntu 18.04

lucasmartins131, 2020-11-23:
I also checked if the simlink was there, and it was. Then run the run ~/git/simulation/installation/install.sh

DanHert, 2020-11-23:
this error - `c++: internal compiler error: Morto (program cc1plus)` often shows up when you run out of RAM. You can open `htop` during compilation to see if you are running out of RAM. You have only 6GB, which is not much, and your CPU has 8 threads, which is a lot, so it will compile 8 packages at once and might run out of RAM.

You can try to expand your swap file - https://bogdancornianu.com/change-swap-size-in-ubuntu/
Or run `catkin build -j1` which will compile only with one thread, so you do not run out of RAM (it will also be much slower)

lucasmartins131, 2020-11-23:
Just to make sure I understood. 
Do I have to re install all again or only do a catkin clean and after a catking build -j1 ?
I already have 4 GB of swaap, and you are right, the RAM memory is going out

DanHert, 2020-11-23:
You probably do not need to reinstall, just try `catkin clean; catkin build -j1`

lucasmartins131, 2020-11-24:
I compiled it all again, this time running one package per time. After I tried to only to open an empy world with roslaunch

lucas@lucas:~/mrs_workspace$ roslaunch mrs_simulation simulation.launch gui:=true
RLException: [simulation.launch] is neither a launch file in package [mrs_simulation] nor is [mrs_simulation] a launch file name
The traceback for the exception was written to the log file

After that I tried to run the tmux_exemple

lucas@lucas:~/mrs_workspace/src/simulation/example_tmux_scripts/one_drone_gps_baro$ ./start.sh

export UAV_NAME=uav1; export RUN_TYPE=simulation; export UAV_TYPE=t650; export WORLD_NAME=simulation; export SENSORS=""
waitForSimulation; roslaunch mrs_uav_status status.launch
lucas@lucas:~/git/simulation/example_tmux_scripts/one_drone_gps_baro$ export UAV_NAME=uav1; export RUN_TYPE=simulation; export UAV_TYPE=t650; export WORLD_NAME=simulation; export SENSORS=""
lucas@lucas:~/git/simulation/example_tmux_scripts/one_drone_gps_baro$ waitForSimulation; roslaunch mrs_uav_status status.launch
waiting for simulation
waiting for simulation
waiting for simulation
waiting for simulation
waiting for simulation
waiting for simulation
waiting for simulation

lucasmartins131, 2020-11-25:
Now I think I am getting closer. I forgot to bash the /devel/setup.bash. Now the empty world is doing ok 
But when I try to spawn UAV it says: 
lucas@lucas:~$ spawn_uav 
[rospack] Error: package 'mrs_simulation' not found

DanHert, 2020-11-25:
Looks like the simulation is not installed properlly. Go to `~/git/simulation` and call `git pull`. Then run the simulation install script again, and recompile the mrs_workspace (`catkin build -j1`, or maybe -j2 to make it a bit faster).

klaxalk, 2020-11-25:
Just small advices: `Error: package 'mrs_simulation' not found` is a standard ROS error. It says, that it cannot find the package from which we are launching a launch file. There may be various reasons for that.
1. You did not source the workspace which contains the package. In our case, this should be done automatically, check if you have the following line in the `.bashrc` file: `source ~/mrs_workspace/devel/setup.bash`
2. The package is not compiled. Then, find the package and try to compile it. The package is in the `~/mrs_workspace/src/simulation/ros_packages/mrs_simulation`

Please, post your logs as formatted text, it is much easier to read then.

lucasmartins131, 2020-11-26:
Guys 
Now I think it is working. I don´t know how to thank you for the pacience. I will now begin studing the topics, and when I have more doubts I will post it in another topic
Best Regards

closed 2020-11-26 by spurnvoj