ctu-mrs/mrs_uav_system #24: Multiple drones transporting a payload (modeling issue)

AlyDiane, 2020-11-25:
Hello MRS team,

We just started using your MRS code for the purpose of our thesis. Our objective is to spawn and control 2 or more drones transporting a payload (example a bar) attached through cables. Right now we are able to spawn multiple drones using your Tmux files, but we don't know how to add/model the cables and payload. Do you have a package or plugin that can help us with this? 

Thank you in advance for your answer!
Best,
Aly

spurnvoj, 2020-11-25:
Hi, check this issue https://github.com/ctu-mrs/mrs_uav_system/issues/14.

AlyDiane, 2020-11-27:
HI, Thank you for your response. We managed to make it work. However, we still have some questions and were wondering if you could advise us:

1) is there a command/way to make the link visible?
2) if not, we thought of using a cylinder to model the cable (connect the drone to the cylinder and then connect the cylinder to the load). For this, the connection should not be in the center of mass. Do you know a way to choose the position of the connection?

Thank you in advance for your help!
Best
Aly

AlyDiane, 2020-11-27:
Hi, thank you for your response. We managed to make it work. However, we still have some questions and were wondering if you could give us some advise:

1) is it possible to make the link visible?
2) if not, we thought of using a cylinder to model the cable (drone connected to cylinder and then cylinder connected to the load). For this the connection should not be in the center of mass (we will also use this feature to connect multiple drones to 1 load). Do you know of a way to make this possible?

Thank you in advance for your help!
Best
Aly

spurnvoj, 2021-01-20:
Hi, sorry for the late response. I missed your comment. Link is visible if it contains property [visual](http://sdformat.org/spec?ver=1.7&elem=link#link_visual).  To specify the points of connection, you can add additional links to both ends of the cylinder and connect to them rather than to the link with the visual property.

Cheers, Vojta

closed 2021-02-04 by spurnvoj