ctu-mrs/mrs_uav_system #25: Initial position of spawned drone

PhilPhal, 2020-12-07:
Hello MRS team,

I am trying to change the initial position of the drone, since flying to the wanted position is less accurate. I found the "__init__.py" file with the initial positions at the directory: "/mrs_workspace/src/simulation/ros_packages/mrs_simulation/scripts/simulation". However, I cannot find where this file is called. Could you help me out?

Thank you in advance.
Best,
Philippe

spurnvoj, 2020-12-07:
Hi,

you can specify the initial position using `--file` argument. See this [session.yml](https://github.com/ctu-mrs/mrs_uav_testing/blob/master/tmux/test_control_gps/session.yml#L23) as an example. The initial position can be loaded from csv or yaml file (see [init_pose.csv](https://github.com/ctu-mrs/mrs_uav_testing/blob/master/tmux/test_control_gps/init_pose.csv)).

Best,
Vojta

PhilPhal, 2020-12-10:
Hi Vojta,

Thank you for your response, it solved the problem!

Best,
Philippe

closed 2020-12-10 by PhilPhal