ctu-mrs/mrs_uav_system #26: Hardware - Use of the RC Receiver Optima SL with a navio2-RaspberryPi3 as flight controller

kefran88, 2020-12-07:
Hello MRS team,

I am currently following your steps (https://ctu-mrs.github.io/docs/hardware/) to build a drone like the ones you are using along with the UAV system. The only difference here is that I am using a navio2-RaspberryPi3 as flight controller. I was wondering which type of ports of the OPTIMA SL you are using and which ones should I use in my case because right now, the transmitter (Hitec Flash 8) and the receiver are bind but my motors aren't responding... The OPTIMA SL's CH1 is connected to the ppm/sbus port of the navio2. 

Could you help me out?

Best,
Frank

DanHert, 2020-12-07:
Hello, to use the sbus output, connect to the SL port of the Optima SL, like in this picture:

[image]

kefran88, 2020-12-07:
Hello Dan, 

Thank you very much for the fast answer. I was also wondering if it is possible to make the drone fly with only the Pixhawk 4  and the Flash 8 controller. I would like to do this preliminary step (to check if everything is allright) before using the Intel NUC10i7FNK as my onboard computer.

Best,

Frank

DanHert, 2020-12-08:
Hello, yes, the pixhawk does not need any outside help, it is quite capable on its own. You can fly mannualy just with the pixhawk, and it also supports altitude hold, position hold with GPS and simple missions. So, yes you can definetly fly without the NUC, in fact I would recommend this step to ensure that the drone is behaving as it should

Dan H.

closed 2021-01-20 by DanHert