ctu-mrs/mrs_uav_system #27: Trackers - how to get access to some physical values

bconvens, 2020-12-08:
Hello,

The Se3Controller's update function takes as input also the UAV's mass, gravity constant, and motor parameters A and B.
How can I get access to these variables inside a custom tracker I want to create?

Thanks!
Best,

klaxalk, 2020-12-10:
Hey, good question. You can't. So far I did not need them in a tracker. However, I acknowledge that they might be used, so I am going to update the tracker interface, so both the trackers and the controllers will get the same info. I will keep you posted.

klaxalk, 2020-12-26:
Hey Bryan, Merry Christmas!! I updated the controller and tracker interface. Now, you can get the following from the `common_handlers`, that are passed to trackers and controllers:
- `common_handlers->motor_params`
- `common_handlers->g` ultimately pass from the [core.launch](https://github.com/ctu-mrs/mrs_uav_general/blob/master/launch/core.launch) to all core nodes
- `common_handlers->getMass()` gets you the current **estimated** UAV mass

The `motor_params` now follows a new format defined in [mrs_lib](https://github.com/ctu-mrs/mrs_lib/blob/master/include/mrs_lib/quadratic_thrust_model.h), pls check [mrs_uav_system#33](https://github.com/ctu-mrs/mrs_uav_system/issues/33).

Sincerely, Tomas

closed 2021-01-20 by DanHert