ctu-mrs/mrs_uav_system #30: Hardware - How to fly or know that the Nuc works

ZakariaLaakel, 2020-12-10:
Dear MRS people,

We have built a F450 drone according to the components you used. The whole drone is assembled and we had questions concerning what should happen next:

- Which script are you using in order to make the drone fly? I think you don't use tmux sessions but I'm not sure.

- Is there a way to know that the NUC is working correctly without the drone to fly? A sort of test or check so we know that the NUC gets all the data.

Kind regards,

Zakaria

DanHert, 2020-12-10:
Before flying autonomously, try to fly manually to test if the drone itself is configured corretly.

Hi, for a normal flight with GPS, use this script - https://github.com/ctu-mrs/uav_core/blob/master/tmux_scripts/just_flying.sh
But you need to setup the proper config in your .bashrc. Especially important are the UAV_MASS, RUN_TYPE (uav) and UAV_TYPE (f450). Once you start the script, check all the tabs if something crashed. Then you can go to the mrs_status tab, to see if everyting is fine. You should see data in the Mavros and Odometry windows (they should be green) and GPS should be present. If you have all of these, you can fly.

ZakariaLaakel, 2020-12-22:
Hi Dan,

thank you for your answer. I tried to setup ROS on the NUC according to the steps mentioned in [https://ctu-mrs.github.io/docs/hardware/px4_configuration.html](url) but I got some problems:

- My PixHawk 4 is not seen as a /dev/ttyUSB0 or /dev/pixhawk device, but /dev/ttyACM0. I tried different USB ports and they all gave me the same name. Is there a way to complete the setup with this name? I changed the the first line of the mavros_uav.launch file from 
	<arg name="fcu_url" default="/dev/pixhawk:921600" />
to 
	<arg name="fcu_url" default="/dev/ttyACM0:921600" />

but I don't know if I have to make changes in other launch files.

- When I launch the just_flying.sh script, am I still supposed to see green tabs on mavros and/or Odom if I have a Pixhawk device with nothing connected to it other than the USB connection between the NUC and the Pixhawk 4.

Thank you in advance!

DanHert, 2021-01-08:
Hello, it does not matter if it is seen as /dev/whatever, just remap it with udev rules to /dev/pixhawk.
Run the command: (replace the /dev/ttyUSB0 with /dev/ttyACM0 or whatever it is connected as)
`udevadm info -p  $(udevadm info -q path -n /dev/ttyUSB0) | grep 'SERIAL_SHORT\|VENDOR_ID\|MODEL_ID'`
and then follow the px4 configuration guide. After you setup the udev rule, just unplug and replug the serial to USB convertor to the computer and you should see /dev/pixhawk connected.

As for the power for the NUC, the NUC usually has an input range of 12V-19V (some models are different), so with a 4S battery you can power the NUC directly. On some drones we use 6S batteries, so a powerful step down converter is used for NUC power.

Sorry for the delayed answer, Dan H.

closed 2021-01-20 by DanHert

ZakariaLaakel, 2021-01-26:
Hello Dan, the configuration is done correctly, thank you very much. I had 2 more questions regarding the battery: 

-You say that you are using a 4S battery to power the NUC. Is it the same battery that powers the other components of the drone? Or are you using 2 4S battery per drone?

- How do you connect that battery to the NUC? Can you maybe show a picture of how you've done it? At the moment we have the entire NUC device in its shield, do we have to dismount it and only keep the board that is inside it?

Kind regards,

Zakaria

DanHert, 2021-01-27:
Hello, the NUC is connected to the same battery which powers the UAV, so we have only one 4S battery. The NUC can take a DC voltage between 12-19 V, so it can be powered straight from the battery, without any convertors. Therefore we simply connect the nuc directly to the battery by soldering its power cable to the distribution board. We also remove the NUC from its box to save weight.

[image]

ZakariaLaakel, 2021-01-27:
This is much clearer now. Thank you very much !

ZakariaLaakel, 2021-02-03:
Hello, sorry I had one last question concerning this topic: When you connect the PixHawk 4 to the NUC through the telem2 port and using this FTDI clone board [https://fr.aliexpress.com/item/4001323460121.html?spm=a2g0o.productlist.0.0.7b83e8dcbCXILT&algo_pvid=1c2070da-7f8f-47ad-a8a0-83f3b4ca9cb6&algo_expid=1c2070da-7f8f-47ad-a8a0-83f3b4ca9cb6-5&btsid=2100bdd016034491508824193e00b8&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603](url).

I see on the pictures of the PX4 configuration section that you do not use all the pins of the telem2 port. From what I can see I deduct you did the following:

- Connecting the first pin of telem2 ( VCC pin) with the VCC pin of the FTDI board;

- Connecting the 4th pin of telem2 (CTS pin) with the DTR pin of the FTDI board;

- Connecting the 5th pin of telem2 (RTS pin) with the CTS pin of the FTDI board.

Are these deductions correct?

DanHert, 2021-02-03:
Hmmm, looking at the picture on our wiki, I notice there is something very wrong with the TELEM2 connector.
The correct connections should be:
PX4 GND -> FTDI GND
PX4 TXD -> FTDI RXD
PX4 RXD -> FTDI TXD
That is all you need.
For some reason the cable in the picture is flipped, I will fix it, thanks for noticing.