ctu-mrs/mrs_uav_system #31: Simultaneous takeoff drones

AlyDiane, 2020-12-10:
Hello MRS team,

I was wondering if there is a way to let two or multiple drones takeoff at exactly the same time? Additionally, is there a way to change the takeoff height of the drones (for example from 2m to 1m)?

Thank you in advance!

Best,
Aly

DanHert, 2020-12-10:
Hello, you can change the takeoff height here - https://github.com/ctu-mrs/mrs_uav_managers/blob/master/config/simulation/t650/uav_manager.yaml
If you are using the t650, if you are using another drone just change it in the according folder.

As for simultaneous takeoff, you have to switch all the uavs to the offboard mode at the same time. So, in your tmux session, comment out all the lines with
`rosservice call /$UAV_NAME/mavros/set_mode 0 offboard'`
And make a new pane with a prepared command to switch all the drones to offboard at the same time. Something like:
`history -s rosservice call /uav1/mavros/set_mode 0 offboard; rosservice call /uav2/mavros/set_mode 0 offboard; ....`
Once your drones are all loaded and ready, put them all into offboard and they will takeoff at the same time.

Dan H.

AlyDiane, 2020-12-17:
Hey,

Thank you for your answer! We created a custom_configs folder in our  folder and added a uav_manager.yaml file there (like it is done in: https://github.com/ctu-mrs/mrs_uav_testing/tree/master/tmux/test_control_gps). In the yaml file, we are able to change the tracker after takeoff but not the takeoff height as this remains on 1.5 m even when we change it to e.g. 3 m. Is the original uav_manager.yaml file inside for e.g the t650 folder (https://github.com/ctu-mrs/mrs_uav_managers/blob/master/config/simulation/t650/uav_manager.yaml), the only way to change the takeoff height as we are not able to do it with our custom configs?

Thank you in advance!

Aly

closed 2021-01-20 by DanHert