ctu-mrs/mrs_uav_system #32: Backward incompatibility: rework of simulation UAV spawning mechanism, Noetic update, WIP

klaxalk, 2020-12-15:
**Work in progress**

- the spawning is now managed by a dedicated "spawner" program
- the "spawner" is launch within the simulation.launch, if you don't use our simulation.launch, then run it manually (todo link to spawner launch)
- spawning is now done via a ROS service call
- use one call for spawning multiple UAVs at once
- the parameters to the spawn command stayed the same, they are now just passed via a ROS service
- [compontes definition in relation to the spawner script]
- the example sessions have been updated [link]
- pulling the simulation repository might need additional actions: remove uncommitted changes from the px4 repository

MvdB123, 2020-12-22:
Hi guys, 

Michiel here. I participated in the summer school of 2019 and am trying to get the seminar task running again (very beginner with ROS, tmux, linux, but experienced with Python on Windows). I therefore first go through the `https://github.com/ctu-mrs/example_ros_packages` as to have a reference.

I changed the session.yml with the following line:
```
waitForSimulation; rosservice call /mrs_drone_spawner/spawn "1 --enable-ground-truth --enable-rangefinder --enable-mobius-camera-front --pos 10 -15 0.3 0.7"
```
I get the following error in the spawn window of tmux: `transport error completing service call: unable to receive data from sender, check sender's logs for details`. Any pointers on how to solve/get a running reference of the uav system? 

Many thanks.

klaxalk, 2020-12-22:
Hey, sorry, I forgot to update the tmux session in the examples. It should be up to date now ([changes](https://github.com/ctu-mrs/example_ros_packages/commit/60f268961ecd6e6e2d163208a6e6e9b856b7393e)). Also, I have patched the 2019 and 2020 summer school repositories ([changes](https://github.com/ctu-mrs/mtsp_planning_task/commit/17afd592391d13b92bbc39de6c79266aeecd3147), [changes](https://github.com/ctu-mrs/uvdar_leader_follower/commit/2fe02c76db56e86713cda2cf339103fd9158aaa9)), with the latest updates, so they should hopefully work now as well. Let me know if you encounter more problems. Sincerely, Tomas

MvdB123, 2020-12-28:
Thanks a lot Tomas. I got a bit further but have some other errors. Potentially unrelated to this one potentially not.

I made some changes to run from python3. (if you'd like to know; i've documented some). Things improved. But I finished with some warning messages when running the summerschool's tmux session. In the control panel the odometry posted something like: callbackGarmin(): No odom_pixhawk. ControlManager posted: not recieving Mavrosstat message. In the Automatic_start panel AutomatStart posted: missing data on topic <path>/ready_to_takeoff. 

I went to this issue and saw the comments about px4. I performed the suggested commands. Then wanted to catkin clean & catkin build the mrs_workspace. I get the following. Any suggestion on what I did wrong? 
[image]

[image]

klaxalk, 2020-12-28:
Hey, this looks like px4 is not compiling properly. Try the following:
1. do `git pull` in the `simulation` repository, this should not output any error, if it does, let me know. Alternatively, you can call `gitman install --force` in the `simulation` repository, to force update the submodules.
2. verify that the commit of `simulation/ros_packages/px4_firmware` by calling `git log` within its folder. It should outptut the following as the first line.
```
commit cbdd1b798bf6f7238aefb8f429bfddae32fd9576 (HEAD, tag: v1.11.2, origin/1.11.2)
```
3. if the compilation error persists, compile it with `--verbose`, it should tell us more.

MvdB123, 2020-12-28:
pull master and checking the commit indeed solves the compilation of this package. Then some others failed. I pulled the uav_core as well. Some other compiles were now failing due to my changes. I pulled those master branches as well (mavros, mrs_rviz_plugins). 

Some errors still remain for mrs_uav_odometry, mrs_uav_controllers and mrs_uav_managers. I pulled the respective master branches, but without effect. Should I checkout a specific commit instead?

klaxalk, 2020-12-28:
Definitely don't pull the respective master branches. In some cases it is fine, but we are not on **master** everywhere. For example in the case of Mavros. As a __normal user__, use gitman to checkout a specific stable combination of commits (`gitman install`, it is called automatically when you do `git pull`). There is an option to "update" everything to the newest version (which might be actually unstable and not working) by calling `gitman update`, but you don't want to do that. More on that topic here: https://ctu-mrs.github.io/docs/software/gitman.html.

So to put everything into "normal state", just do `git pull` in the **uav_core** and **simulation**. You can enforce the checkout of the stable commits by `gitman install --force` if gitman objects (this can happen if you have local changes in the submodules or if your .gitman.yaml file has changed, e.g., due to `gitman update`).

MvdB123, 2020-12-29:
I tried your solution, but it kept failing. I then removed the simulation and uav_core repostirories from my git folder, cloned them, installed them and build them. Still the same errors. This morning I tried resetting fully: delete the mrs_workspace, delete the git folder, delete the repos in the workspace folder. Furthermore I deleted what-ever I could find in my home folder that was related to the uav system. 

Then I tried to freshly install the uav system. But I still get these errors. Should I go for an even more rigorous clean up?
[image]

I also have a lot (231) warnings in my console output. Things like these: 

1.
[image]

2.
[image]

3.
[image]

klaxalk, 2020-12-29:
Hi, so there are obviously multiple problems with your setup. You had the right idea with removing everything and installing it again, but we have to probably go even further.

1. It looks like you have a broken dependency chain in `apt`. This needs to be fixed before any automatic installation script can get to work. The "gazebo9 is not going to be installed" message means, you can not simulate, because the simulator is not going to be installed. The easiest solution is to remove everything which has to do with "gazebo9" and install it freshly.
```bash
sudo apt remove "gazebo9*"
sudo apt remove "libgazebo9*"
sudo apt remove "ros-melodic-gazebo*"
```
I also recommend calling `sudo apt upgrade` just to make sure everything is fine and up to date. If this fails, then fix missing packaged, etc.
Fresh installation of Gazebo is done by the `install.sh` script in `simulation/installation`.

2. When you did the fresh compilation of `mrs_workspace`, you got the "c++ killed signal terminated program c11plus" error. This usually means that you do not have enough RAM to compile all the stuff in parallel (vaguely speaking, you need approx. 2 GB ram for each CPU thread). Fixing this is easy, just compile with, e.g., two threads: `catkin build -j2`. This is usually necessary only during the first compilation where all the packages are clean. It also helps to just rerun the compilation again.

3. after you fix the installation, start with something simple, rather than the MTSP task. I recommend running the `simulation/example_tmux_scripts/one_drone_gps` sessions. That is the "minimal example". If the UAV takes off then the system is fine. Any more problems will be probably with the MTSP task repository.

It is often easier to just run the `install.sh` in the `mrs_uav_system` repository. The installation should stop whenever there is a problem. If it stops and says something like `... command failed with exit code ...`, then copy the surrounding text and post it here. Don't try to run the simulation (it won't work with unfinished installation) and don't try to fix it manually. However, in your case, we need to do some manual fixing, e.g., due to the broken dependencies.

lidiaxp, 2021-03-20:
I tried run the example on tmux, `just_flying.sh`. The UAV scenario and UAV appear but don't takeoff. The second terminal (Gazebo with command `'Gazebo' "waitForRos; roslaunch mrs_simulation simulation.launch world_name:=grass_plane gui:=true`) return this error: 

```
process[uav1/sitl_uav1-3]: started with pid [72168]
INFO  [px4] PX4 daemon already running for instance 1 (Success)
[uav1/sitl_uav1-3] process has died [pid 72168, exit code 255, cmd /home/lidiaxp/mrs_workspace/devel/lib/px4/px4 /home/lidiaxp/mrs_workspace/src/simulation/ros_packages/mrs_simulation/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -i 1 -w sitl_uav1 -d __name:=sitl_uav1 __log:=/home/lidiaxp/.ros/log/9931fb48-89a2-11eb-b7d5-a7829f5bef3d/uav1-sitl_uav1-3.log].
log file: /home/lidiaxp/.ros/log/9931fb48-89a2-11eb-b7d5-a7829f5bef3d/uav1-sitl_uav1-3*.log
all processes on machine have died, roslaunch will exit
[INFO] [1616261829.566427, 6.284000]: [DroneSpawner]: Firmware for uav1 started!
[INFO] [1616261829.623215, 6.284000]: Spawn status: SpawnModel: Successfully spawned entity
```

And others terminal still waiting for odometry and control.

ZakariaLaakel, 2021-03-20:
Hi,

When installing your linux setup and software on a brand new computer I am not able to compile the mrs_workspace. I keep getting the following error when compiling:

[image]

Any tips on how to fix this?

ZakariaLaakel, 2021-03-21:
Hi,

I've been using your software for a few months now. I never got big issues when it comes to installation but I tried to install your linux setup and software on a brand new computer Today and I was not able to compile the mrs_workspace. I keep getting the following errors:

[image]
[image]

Any tips on how to fix these?

klaxalk, 2021-03-22:
Hey, since you are running the example session (not your own script) and this issue is regarding backward incompatibility for personal custom tmux script, I suggest moving this into the discussions. But this can happen for many reasons. Is our system up-to-date and does the _mrs_workspace_ compile fully without errors?

lidiaxp, 2021-03-22:
Yes, I installed MRS system last week and `catkin build` don't returned errors.

spurnvoj, 2021-03-22:
Hi, this error tells you that the pixhawk instance is already running. So call `killall px4` in terminal to kill it manually or use this key sequence https://ctu-mrs.github.io/docs/FAQ.html#how-to-stop-the-simulation for the clear shutdown of everything in the tmux session

lidiaxp, 2021-03-23:
Thanks, it worked