ctu-mrs/mrs_uav_system #37: Enabling Realsense Camera

lpmrz, 2021-01-30:
Hello,

I am trying to use PointCloud2 data for 3D mapping purposes. What I have done so far:

- I saw that it is being published under the topic /uav1/os1_cloud_node/points. However no node seems to publish it. I tried to enable Realsense camera by adding the following parameter (--enable-realsense-front-pitched)to the session.yaml. If the drone is supporting it, simulation launches without errors. But Realsense came does not appear to be working.

Here is the status window:

[image]

I think I am missing something. Do you have any recommendations?

DanHert, 2021-02-01:
Hello, the problem is not that the realsense is not working, but the status window is checking a wrong topic. I fixed the status node, so now it should display the realsense correctly. I also fixed the duplication of the Garmin topic.

Pull uav_core and recompile and the status should be displayed correctly. But the realsense was working all along.

lpmrz, 2021-02-01:
Thank you for your help. I pulled the uav_core and did a catkin build. But now there is this error:
[image]

DanHert, 2021-02-03:
Hi, I dont have any problems with the simulation, so probably just going to ~/mrs_workspace and calling
`catkin clean`
`catkin build`
should fix this.

lpmrz, 2021-02-04:
Thank you very much. I cleaned and built again, now Realsense and the status are working perfectly. Unfortunately I can't obtain any PointCloud2 data. I type _rostopic find sensor_msgs/PointCloud2_ and it returns:

- /uav1/bumper/lidar3d_processed
- /octomap_point_cloud_centers
- /uav1/os1_cloud_node/points

When I type _rostopic echo /uav1/bumper/lidar3d_processed_ it publishes nothing. Others return empty too. Do I have to do something extra for those topics to start publishing?

Thank you again for your time and interest.

DanHert, 2021-02-04:
Hello, our simulated realsense does not produce any point cloud messages. It produces a depth image on a topic
`/$UAV_NAME/rs_d435/aligned_depth_to_color/image_raw`
You can either modify the simulation plugin to publish point clouds, or use this package - http://wiki.ros.org/depth_image_proc
to convert the depth images into point clouds.

WingRS, 2021-03-09:
Hi all,
By my opinion Dan has provided a nice explanation of the PCL, I am closing the issue. 
Thanks to all. :)

closed 2021-03-09 by WingRS