ctu-mrs/mrs_uav_system #38: Question about lidar sensor.

Gaoyuan-Liu, 2021-02-04:
Hey Mrs team,
I notice you integrated a laser scanner called "rplidar" in the drones (F550), I'm recently using it to detect obstacles. 
My question is what is the real-world lidar sensor you are using onboard? Since we are trying to build a counterpart real-world drone as the simulation.

Thanks for your help! : )

Gaoyuan

DanHert, 2021-02-04:
http://letmegooglethat.com/?q=rplidar

But jokes aside, it is a rplidar A3 - https://www.slamtec.com/en/Lidar/A3

Gaoyuan-Liu, 2021-02-04:
Hey Dan,
Sorry about the silly question. Yes, I also googled that but wanted to make sure ;)
And could you please also give some information about what sensor/camera you are using currently to achieve obstacle avoidance, I also saw a long list in the urdf file about sensor/camera, but which one (or combinations for data fusion maybe) you will recommend using on a real-world drone (let's say F450 this time)? And what's the consideration of your preference?

Thanks again!

DanHert, 2021-02-04:
On our F450s, we use the rplidar for obstacle avoiodance. It is only a planar sensor, but that is enough in some enviroments.
In more cluttered environments, we are using 3D lidars (mostly Ouster OS0), complemented by stereocameras (Intel Realsense).

Dan H.

Gaoyuan-Liu, 2021-02-25:
Hey Dan,
I'm trying to use octomap to do mapping and path planning.
Is there any camera/sensor plugin written in urdf can generate `/sensor_msgs/PointCloud2` data?

Thanks for your help.

DanHert, 2021-02-26:
Hello!
You can use a 3D lidar in the simulation (add "--enable-velodyne" into the spawn command in simulation session file). It will output PointCloud2 directly.
Alternatively, you can use a simulated stereocamera ("--enable-realsense-front" other orientations available) which will give you a depth image, but that can be converted to PC2 with http://wiki.ros.org/depth_image_proc
The LaserScan data from rplidar can be also converted to PC2, with http://wiki.ros.org/laser_geometry

cheers, Dan H.

Gaoyuan-Liu, 2021-02-26:
Hey Dan,
Thanks a lot for your response!
I just tried "--enable-velodyne", indeed, it can give a PC2 topic but there is an error in rviz:
"Message
Data size (333750 bytes) does not match width (0) times height (1) times point_step (25). Dropping message."

Maybe I also need some configuration before I use it directly.

Thanks!

Gaoyuan-Liu, 2021-02-26:
Hey Dan,
Thanks a lot for your response!
I just tried "--enable-velodyne", indeed, it can give a PC2 topic but there is an error in rviz:
"Message
Data size (333750 bytes) does not match width (0) times height (1) times point_step (25). Dropping message."

Maybe I also need some configuration before I use it directly?

Thanks!

DanHert, 2021-02-26:
Yeah, looks like we have some bug there. Switch to "--enable-ouster" instead of velodyne, that one should work normally.

Cheers, Dan H.

WingRS, 2021-03-09:
Hi all,
@DanHert  have you been able to test the velodyne in RviZ?

petrapa6, 2021-03-09:
Hi, 

velodyne was fixed in [297f0ba](https://github.com/ctu-mrs/simulation/commit/297f0ba47770c592b669b1ece5b3f18f2c37a1e4).

closed 2021-03-09 by petrapa6

ZhiWei20011, 2021-06-30:
Hi, after adding the "--enable-realsense-front" to the simulation command, how can I see the image from rviz? Do I need to change some settings in rviz? Appreciate any answers. Thank you.

ZhiWei20011, 2021-07-02:
Hello??

petrapa6, 2021-07-02:
Hello,
you definitely need to add manually all the data you want to visualize in RVIZ on your own. I believe using the RVIZ GUI is intuitive and straightforward, and does not need any more explanation. I can point you to [http://wiki.ros.org/rviz](http://wiki.ros.org/rviz) for more information.

If you have an unrelated question to this issue, please either open a new [issue](https://github.com/ctu-mrs/mrs_uav_system/issues) or use [discussions](https://github.com/ctu-mrs/mrs_uav_system/discussions). Closed issues and PRs are no place for unrelated questions.

Best, Pavel