ctu-mrs/mrs_uav_system #39: About flying a custom bild UAV

maxhofidrone, 2021-02-09:
Hi,
I set up a Holybro X500 with a Raspberry Pi 4 as onboard computer. I could already verify a correct connection to the Pixhawk itself (mavros). Furthermore I followed your [Pixhawk config/setup](https://ctu-mrs.github.io/docs/hardware/px4_configuration.html)
In following comment by @DanHert in this [issue](https://github.com/ctu-mrs/mrs_uav_system/issues/30#issuecomment-742483814) it is described to update the .bashrc. UAV_MASS (to be measured before launch) and RUN_TYP as uav are clear to me. But I have a question about the UAV_TYP: The X500 is not given in the options- can I just add is nevertheless? Or are there  additional settings to be made in different files (like adding a new UAV in simulation)?

Furthermore I would have a question to the custom config files and the .world file for real UAV flights:
Can I just add custom configs as in the simulation? For example, I don't have any rangefinder on the UAV. The altitude should just be estimated by the baro of the Pixhawk (= custom odometry config).
To the world file- The example world files include following option: `use_utm_origin`- Setting this to false would choose the starting point as origin of the local system right?

Best regards,

Max

DanHert, 2021-02-09:
Hello,
the UAV_TYPE is used for custom configs for the various managers (gains, constraints, control) located here - https://github.com/ctu-mrs/mrs_uav_managers/tree/master/config/uav
If the type is not defined, the default configs will be used - https://github.com/ctu-mrs/mrs_uav_managers/tree/master/config/default

For real world flights, we no longer use tmuxinator, but more plain tmux scripts. Examples can be found here - https://github.com/ctu-mrs/uav_core/tree/master/tmux_scripts
there is even an example with custom configs (just_flying_hector_opflow). So you can supply a custom config for odometry there. Also make sure to set PIXGARM (in .bashrc) to false and remove any mentions of garmin from the SENSORS variable in .bashrc. There may however be problems with this setup, as we never fly just with baro, if the baro estimator drifts "under ground" - the Z goes negative, the drone may not takeoff.

And the point about use_utm_origin is correct, disabling it will treat the position where the drone is turned on as 0 0.

There could be other issues when running on the raspi 4, @klaxalk will add some more comments.

Dan H.

klaxalk, 2021-02-09:
Hey, we have not yet flown with Raspi4 and I expect there could be some issues regarding the real-time performance of the system. For the first tests, I would recommend decreasing the rate of the MPC Tracker loop in the [mpc_tracker.yaml](https://github.com/ctu-mrs/mrs_uav_trackers/blob/master/config/default/mpc_tracker.yaml) from 100 Hz to 10 Hz (it can fly with this just fine unless you do really aggressive maneuvers).
```yaml
mpc_rate: 10.0 # rate of MPC calculation, >= 10 Hz
```

Before you fly, definitely check the rate and stability of the main odometry/uav_state at `odometry/uav_state`. Normally it is 100 Hz. We can fly with less (e.g., 30 Hz). But it would be better to set it deliberately rather than to experience dropouts from 100 Hz during the flight. Let me know if you would think it needs to be changed.

Moreover, make sure the output of the core.launch does not contain any weird warnings or errors. Watch for warnings about "odometry hiccups" and the "rate of Mavros state".

I recommend compiling your workspace using the "release" profile (`catkin profile set release`) which will enforce `-O3` optimization that should help with the performance.

What motor constants do you have? How did you obtain them?

maxhofidrone, 2021-02-10:
Thank you very much for your comments!

- Baro: I could add a TFminiPlus rangefinder. I originally wanted to avoid it, as we plan to fly higher then its range (~10m) and also in a hilly environment. Using a rangefinder here already caused some problems in the [simulator](https://github.com/ctu-mrs/mrs_uav_system/issues/18)
- Custom configs: yes I saw this example. I just want to set the custom configs for the height estimator and the .world file. but it is now clear for me
- MPC tracker loop: I will try this- in general, can this loop be chosen as any rate between 10-100Hz? So I could test a bit what works best. Are there any other tracking/control rates which might need to be adopted in the same way?
- odometry/uav_state: I started it once and could not see something you mentioned. Within the status window everything was green- But I will check it again. Same for the potential errors of core.launch
- compiling: this profile you mentioned will only have an effect for the compiling itself right? Until now I just compiled with `catkin build -j2` to avoid freezing of the system.
- motor constants: I don't have any nor did I know that I need them for flying a real UAV. I thought they are only needed within the simulation to simulate the UAV realistically. We bought the standard kit of the [X500](https://www.banggood.com/Holybro-X500-Pixhawk-4-500mm-Wheelbase-Frame-Kit-Combo-2216-880KV-Motor-1045-Propeller-for-RC-Drone-p-1588063.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=DE&utm_content=minha&utm_campaign=minha-de-en-pc&currency=EUR&cur_warehouse=CN&createTmp=1&ID=529053&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=frank-ssc-de-css-all-newcustom-ncv90-0623&ad_id=443949120684&gclid=Cj0KCQiAgomBBhDXARIsAFNyUqP44nJxnt9YMWiZ_SC2hTZpV5-pCW5RaReoVb-6fQ2EKYqQFp--kQAaAn42EALw_wcB) with its standard [motors](https://shop.holybro.com/motor2216-880kv-1pc_p1154.html). They include some parameters, but I don't know if this is enough information for the motor constants required by MRS:
[image] If not- how can I obtain these parameters? Also where are the motor parameters saved for the existing models?

Thanks for your help again! I hope to make some first test flights on Friday

klaxalk, 2021-02-12:
Hey, sorry for the late answer. `git pull` in the uav_core please.

> * Baro: I could add a TFminiPlus rangefinder. I originally wanted to avoid it, as we plan to fly higher then its range (~10m) and also in a hilly environment. Using a rangefinder here already caused some problems in the [simulator]
(https://github.com/ctu-mrs/mrs_uav_system/issues/18)

Yep, in the case of a hilly environment, just stick to BARO.

> * MPC tracker loop: I will try this- in general, can this loop be chosen as any rate between 10-100Hz? So I could test a bit what works best. Are there any other tracking/control rates which might need to be adopted in the same way?
Yes, you can set it to anything in this range. Feel free to experiment. Just beware, the MPC tracker does not consume CPU resources when the UAV sits on the ground. The main load will appear only during the flight when the tracker is active. So definitely start with 10 Hz. But as I said, you will probably not see the difference unless you do some very aggressive maneuvers.
> * odometry/uav_state: I started it once and could not see something you mentioned. Within the status window everything was green- But I will check it again. Same for the potential errors of core.launch

Great, just keep an eye on the errors and warnings.

> * compiling: this profile you mentioned will only have an effect for the compiling itself right? Until now I just compiled with `catkin build -j2` to avoid freezing of the system.

The profile influences how much is the final binary "optimized". With higher optimization, the compile-time is longer but the run time is usually faster.

> * motor constants: I don't have any nor did I know that I need them for flying a real UAV. I thought they are only needed within the simulation to simulate the UAV realistically. We bought the [...]

We have not yet flown with X500 autonomously (but we will soon), so I had to create new configs for it just now. I also created the thrust curve according to your table, so hope you got it right :-). To "properly" use the drone's profile, set the following into the .bashrc:
```bash
export RUN_TYPE="uav"
export UAV_TYPE="x500"
export PROPULSION_TYPE="air2216"
```
This will make the nodes automatically load the "UAV Model's" custom configs (from `config/uav/x500/...` subfolders). I roughly set the control gains and other parameters according to our F450 which should be very similar. In any case, all the attitude gains are slightly softer than what the fine-tuned gains probably will be.

The motor parameters are stored in the [mrs_uav_controllers](https://github.com/ctu-mrs/mrs_uav_controllers) repository under the [folder](https://github.com/ctu-mrs/mrs_uav_controllers/tree/master/config/uav/x500) related to the particular UAV model. The parameters are calculated by fitting a quadratic thrust model on the thrust-throttle data you provided, see [this matlab script](https://github.com/ctu-mrs/uav_core/blob/master/miscellaneous/thrust_constants/motor_thrust_curve_estimation/air2216_880kv_T1045.m) I just made.

Make sure you set the initial takeoff mass properly into the .bashrc as well:
```bash
export UAV_MASS="3.0"
```

maxhofidrone, 2021-02-15:
Hi @klaxalk 
we had our first- partly successful- flight today.
First of all, we could not see errors regarding the mavros and only one "hiccup" (right at the beginning):

```
.
.
.
[ INFO] [1613400351.108509807]: [ControlManager]: tracker: 'NullTracker', controller: 'EmergencyController', mass: '1.90 kg', disturbances: body [0.00, 0.00] N, world [0.00, 0.00] N
[ WARN] [1613400351.190807881]: [ControlManager]: motors are off
.
.
.
```
But we also needed to set the` escalating_failsafe/rc/enabled` to `false`. We are using a [Futaba T8J](https://futabausa.com/product/8j/) RC, where the channel 7 is mapped to a turning knob  resulting in following warnings in case above mentoned parameter was set `true`:

```
.
.
.
 WARN] [1613396672.807743223]: [ControlManager]: triggering escalating failsafe by RC               
[ WARN] [1613396672.807978015]: [ControlManager]: escalating failsafe triggered                      
[ WARN] [1613396672.816529923]: [ControlManager]: motors are off                                     
[ WARN] [1613396672.907993379]: [ControlManager]: escalating failsafe triggered                      
[ WARN] [1613396673.026158449]: [ControlManager]: escalating failsafe triggered   
.
.
.     
```              

For flying itself, we chose the standard just_flying.sh, and just deleted the entry about nimbro. After many try-and-error attempts we managed to take off with following procedure:
1. plug in UAV battery and press the Pixhawk safety button (the RasPi was already running supplied with a powerbank)
2. ssh to the RasPi and starting just_flying.sh
3. Waiting until mavros/odom are "green": [image]
4.  Arming the UAV via QGC
5. switching the offboard switch (channel 5) into offboard mode
6. UAV took off

is this the intended approach? I thought, that the MRS system itself request offboard mode and then take off itself. Furthermore the UAV started to fly circular rounds around the takeoff place directly. It looked "controlled"- even if I expected from just_flying.sh the UAV just to take off and remain in 1.5m. As it started snowing, we couldn't reproduce the circular flying, but I will test it again later this week.

Best regards,

Max

DanHert, 2021-02-15:
Hello, the approach is correct, the MRS system is not set up to transition to the offboard mode on its own. This is always done by the safety pilot, for safety resasons. Once the safety pilot switches to offboard mode, the automatic start node will perform the takeoff after a few seconds.

The circular flying you are describing can be result of bad magnetometer calibration, or interference. If you fly with GPS, the heading of the UAV in the global GPS frame (== north) has to be known (obtained from magnetometer). If the magnetometer is not calibrated properly or there is interference, the measured heading is different from the real heading, which leads to the whirpooling motion you observed.

maxhofidrone, 2021-02-16:
Hi @DanHert ,
thanks for your answer!

I also thought of some miscalibration issue. I'll recalibrate everything before the next flight.

Is there any way to define the end-altitude of the automatic_start.launch? E.g. I want the UAV to climb to 5m befor switching the trackers? I could not find an information in the just_flying.sh or automatic_start.yaml. My other option would be just to request a "goto" after finishing the take off. 

Best,

Max

DanHert, 2021-02-16:
The takeoff height can be changed in the uav_manager config:
https://github.com/ctu-mrs/mrs_uav_managers/blob/master/config/default/uav_manager.yaml
(or in your custom equivalent) - section takeoff -> takeoff_height.

Cheers Dan H.

WingRS, 2021-03-09:
@maxhofidrone  it would be great if you could make a short description of your experience and difficulties that you have faced while building and flying a custom UAV using our setup, so we will have a better guide for future users.
Could you write something like that ?
And maybe a picture of the UAV for us? :)

maxhofidrone, 2021-03-11:
sure I'll write some points togehter and post it here.

maxhofidrone, 2021-04-14:
Sorry for the long delay, here is a short summary about our test with a X500 UAV:

- UAV frame: Holybro X500
- Onboard-PC: Raspberry Pi 4, 8GB RAM
- RC: Futaba T8J

[image]

The construction with the threaded rod is not optimal, but was easy and quick to set up. At the moment the Raspberry Pi is powered by a power bank to keep it running during changing the UAV battery.

Problems we ran into:
- compiling MRS: to avoid to get stuck during catkin tool we need to compile with the -j2 option. Additionally we use  catkin profile set release as @klaxalk proposed
- setting up the RC: We are using the Futaba T8J. Channel 7 was assigned to a rotary knob on this RC. Therefore, it always changed things on the UAV. We needed to turn it of [here](https://github.com/ctu-mrs/mrs_uav_managers/blob/12f1b386690a8e57e573916ec760a61b0ed8b61f/config/default/control_manager.yaml#L92)
- running the just_flying.sh script: we had several crashes at the beginning. With investigation of the tmux output of the control window we found the cause in the ROS bag recording nearly all topics. Deleting 'Rosbag' 'waitForOffboard; rosrun mrs_uav_general record.sh fixed this issue.
- Flying: It was unclear for us at the beginning how to take off:
1. starting the just_flying.sh script with the automatic take off
2. waiting until onboard pc autoarmed the uav
3. switching channel 5 to onboard mode
-->Take off

Up next:
•	Testing the RTK integration described in [issue 34](https://github.com/ctu-mrs/mrs_uav_system/issues/34).  It relates to the other issues here regarding the use of RTK (via serial interface etc.)
•	Find a suitable wifi antenna for the Raspberry Pi (it is hard to find an antenna for RasPi with Ubuntu 18 and always involve compiling drivers...)

In gerneral the MRS system is very impressive and helps a for the progress in our own project- thanks for sharing it!

closed 2021-04-14 by maxhofidrone