ctu-mrs/mrs_uav_system #40: error in build

mrshab72, 2021-02-09:
hi 
when i try to build the project T i have an error :
in  mrs_uav_trajectory_generation 
#include <mrs_uav_trajectory_generation/drsConfig.h>
drsConfig.h not exist!
please help me 
tanks for you

DanHert, 2021-02-09:
Hello, I am not able to replicate the problem, so probably going to `~/git/uav_core` and calling `git pull`
And then rebuilding the workspace:
`catkin clean`
`catkin build`
inside mrs_workspace should fix the problem.

klaxalk, 2021-02-09:
Hey, **mrs_uav_trajectory_generation/drsConfig.h** is header generated by the Dynamic Reconfigure Server. It is created during the compilation automatically. Are you compiling it using the catkin tool?

WingRS, 2021-03-09:
The issue has been opened for around 28 days, I think @mrshab72  has figured it out, (I can bet that the usual ```catkin clean && catkin build``` worked).
I am closing the issue @klaxalk ?

closed 2021-03-09 by WingRS