ctu-mrs/mrs_uav_system #41: Problem running the Optical Flow example scripts

lkova, 2021-02-12:
I am unable to takeoff with a drone in any example that uses optical flow as a primary sensor. The problem is that odometry is not initialized and the optical flow velocity is not published.

The following is the terminal output from the script example_tmux_scripts/one_drone_optic_flow, but it happens with other optical flow examples as well.

[image]

What can be the problem with this and how can I solve it?

Thanks!

DanHert, 2021-02-12:
Hello!
Did you actually clone and compile the optic flow repository? - https://github.com/ctu-mrs/mrs_optic_flow
It will not be cloned by the default install script.

So clone the optic flow repository, run the install script from the install folder, link it to you workspace and compile.

Dan H.

lkova, 2021-02-15:
Thanks!
That was the solution to my problem.

Btw, thanks for sharing this with the community, it has been of great help.

Luka

closed 2021-02-15 by lkova