ctu-mrs/mrs_uav_system #43: Flying with the RTK Emlid Reach M2

ZakariaLaakel, 2021-02-23:
Hello MRS_group,

I had some questions concerning the usage of the RTK gps of Emlid. We have to use the Emlid RTK reach M2 in combination with the LoRa radio from Emlid as well:

- Is the GPS module from the PixHawk 4 still useful when the RTk and LoRa are used? In other words should it still be present on the drone via the PixHawk4?
- Considering the 2 or 3 components from above, we want to reduce the distance between the GPS sensor and the center of rotation of the drone. Which component(s) should be placed at the center of the drone? The GPS module of the PixHawk 4 (if it should be used)? The Reach M2? LoRa Radio? Multiple/All of them?
- If all components should be used and placed as central as possible, are there any risks of interference between the components when placed too close to each other?

Thank you for the answer and kind regards,

Zakaria

DanHert, 2021-02-23:
Hello, we were flying with the RTK just today, and the performance was very good.

The GPS module from pixhawk is still pretty much mandatory, for several reasons:
- The regular GPS serves as a backup
- It has a magnetometer integrated into the receiver, which is necessary for GPS/RTK flights. Pixhawk has also a magnetometer, but it is a crappy one and it is usually close to motor cables, which equals lots of interference.
- It has the arming button integrated.

The coordinates you get from the  RTK correspond to the position of the antenna, so only the antenna should be in the center of the drone, then a simple Z offset will get you to the center of the drone frame. Other components can be mounted anywhere you want. The Pixhawk GPS can be put right beside the RTK antenna, it does not have to be in the center of the drone. Generally, you want to keep the antennas as far as possible from any other electronics, especially high bandwidth devices like USB3.0 cameras or 3D lidars can jam the GPS signals completely. But the antennas themselves can be mounted right next to each other.

We connect the RTK straight to the computer with a USB cable, and send NMEA messages. mrs_serial package can then read the messages out. If you want to use this method, make sure to do "git pull" in uav_core (you need this commit - https://github.com/ctu-mrs/mrs_uav_odometry/commit/4b9e09ac15c092d29d8e7f7b13ecc87fea4c7d25 ).

Dan H

Dan H.

kefran88, 2021-02-25:
Hello Dan Hert, 

I am using the same configuration as above. I was wondering if you guys at MRS used the M2 antenna. In Emlid Documentation (https://docs.emlid.com/reachm2/antenna-placement/) , this one is mentioned several times but in your drones's pictures I don't see it : are your drone able to fly without it ? 

Kind regards, 

Frank

DanHert, 2021-02-26:
Hello,
yes we are using the M2 antenna - https://store.emlid.com/product/multi-band-gnss-antenna/

I am attaching a photo of our drone which flew with the RTK. We are using a bit of aluminium foil for additional shielding, as the antenna is right above the computer. If you put the antenna further away, the shielding is prob. not required. 

[image]

Cheers, Dan H.

kefran88, 2021-03-01:
Thank you Dan Hert, 

I see things clearer right now. 

I am currently following the documentation of Emlid to bind the Reach M2 to the Reach RS2, but I would like to know : what is your routine before a flight ? Especially in order to ensure that everything is okay for these 2 components. 
I have read this (https://ctu-mrs.github.io/docs/hardware/tersus_RTK_GNSS.html) but is there anything else I should be aware of for the first flight with your drone configuration ? 

Have a nice week,

Kind regards,

Frank

DanHert, 2021-03-01:
The Tersus RTK section on the wiki is outdated, we should probably remove it.
When flying with the RTK, we just set up the basestation, let it average the position for a few minutes and thats it. 
Once the RTK reports L1_INT on the drone, the RTK fix is achieved.

Dan H.

WingRS, 2021-03-09:
Thanks to all, I think the issue is resolved, I am closing it :)

closed 2021-03-09 by WingRS