ctu-mrs/mrs_uav_system #44: RTK deactivate Range finder in vertical direction

AlyDiane, 2021-02-24:
Hello MRS team

We are using the RTk gps for our simulation and noticed that it uses the range finder in the vertical direction. Would it be possible to use the RTK gps in all directions?

Best regards
Aly

DanHert, 2021-02-24:
Hello, the RTK altitude estimator is currently not implemented in the system (to my best knowledge), but it will probably be an easy
addition, what do you think @petrlmat?

petrlmat, 2021-02-25:
It should be possible. I will take a look at that.

petrlmat, 2021-02-25:
So the RTK altitude estimator is now available. Pull `uav_core` and put `altitude_estimator: "RTK"` and `altitude/use_rtk_altitude: true` into your odometry config to enable it.

closed 2021-02-25 by petrlmat

AlyDiane, 2021-03-01:
Hello,

Thank you for the update. Is it possible to make the enable this feature without changing your code by e.g. changing it in our own tmux script or config files?

With kind regards
Aly Diane

AlyDiane, 2021-03-16:
Hello,

Sorry for the late reply. We tested the the RTK in the vertical direction but it seems to not work. You can find the video of the test in the attachment.

Best regards
Aly

https://user-images.githubusercontent.com/68328413/111337998-db64a080-8676-11eb-8ad7-49ac20fc7672.mp4

petrlmat, 2021-03-16:
Hi,
I forgot to mention one more step, sorry for that. Please add the `--enable-ground-truth` argument to the spawn command, e.g., `rosservice call /mrs_drone_spawner/spawn "1 t650 --enable-ground-truth"`.

After that, the RTK corrections will be used and you should see this line in the `control` tab of the tmux session:
`[Odometry]: Fusing RTK position`

Regards,
Matej

AlyDiane, 2021-03-17:
Hi,

It works! thank you!

Best regards
Aly