ctu-mrs/mrs_uav_system #45: Using the software to simulate fully actuated hexarotor dynamics

kefran88, 2021-03-01:
Hello CTU team,

I have been using your software for a while and I would like to use it to study the dynamics of fully actuated hexarotor. 
I have noticed that in the models there is already an hexarotor f550, but to make it fully actuated, I need to be able to tilt the rotors. Is it possible to do it directly in the xacro file ? Or do I need some extra files and build another xacro file ? 
 
Best regards,

Frank

DanHert, 2021-03-01:
Hello, you can just modify the .xacros to tilt the rotors:
[image]

But we use PX4 as a low level controller, and to my knowledge it does not support fully actuated platforms. But you can maybe try something like this - https://github.com/castacks/PX4-fully-actuated

Cheers, Dan H.

WingRS, 2021-03-09:
Hi @kefran88,
Since Dan has answered your question (looks crazy if you ask me), I am closing the issue, if you have more questions - feel free.

closed 2021-03-09 by WingRS