ctu-mrs/mrs_uav_system #46: Subscribe to topic from Gazebo

PhilPhal, 2021-03-04:
Hi MRS team,

We need to subscribe to a topic from Gazebo to control the position of our payload. We suppose that you subscribed to the uav_state topic in order to have the position of the uavs. Could you explain where and how you implemented this in your code?

Thank you in advance.

Best  regards,
Philippe

DanHert, 2021-03-08:
Hello, you mean how is the position of the UAV in the uav_state topic calculated?
It is deduced from the simulated Accelerometers/Gyros/Barometers on Pixhawk, and fused with a source of localization, which can be GPS, Lidar SLAM, optic flow, VIO etc.

Dan H.

WingRS, 2021-03-09:
Hi @PhilPhal ,
Since Dan has given a fool answer to your question, I'll close this issue today, or maybe you still have some question regarding the ```uav_state``` ?

WingRS, 2021-03-09:
Sorry, closed it by mistake :) @PhilPhal  waiting for you :)

WingRS, 2021-03-11:
As no answer appeared, I am closing this issue, thanks to all :)

closed 2021-03-11 by WingRS