ctu-mrs/mrs_uav_system #48: Communication between arduino and ROS using mrs_serial package

RaphaelNetels, 2021-03-16:
Hello MRS team,

For my project I need to add a rotary encoder that is connected to Arduino. I tried using the mrs_serial package by adding in my Arduino code the function that will send a 16 bit integer through the serial line, using the Baca protocol (https://github.com/ctu-mrs/mrs_serial). 

Once this is done how should I call the serial node, by making my own launch file? If yes, do I need to add this in the session.yaml?

Thank you in advance for your response!

Best regards,
Raphaël

DanHert, 2021-03-16:
Hello,
you need to remap the arduino serial device as something like /dev/arduino, you can find out how to do that here - https://ctu-mrs.github.io/docs/hardware/px4_configuration.html (Section ROS setup, it is written for Pixhawk, but it should be the same for arduino)

Then change the launch file uav.launch (or copy it and make you own) in mrs_serial (device name should correspond to /dev/arduino, or whatever you changed it to, and baud rate has to be set correctly). Then you can launch the node and it should work :D

Cheers, Dan H.

RaphaelNetels, 2021-03-17:
Hello,

Thank you very much for your response, it works!

Raphaël

closed 2021-03-17 by RaphaelNetels