ctu-mrs/mrs_uav_system #49: Flight : Takeoff issues

kefran88, 2021-03-19:
Hello MRS team, 

I am moving towards my first flight but I have several problems in my set-up when starting the just_flying.sh. 
Our drone setup is the same as you, with RTK and Pixhawk connected in USB. 

1 : all the control and odometry nodes seems to crash. We set up our ~/.bashrc file as required (ODOMETRY_TYPE = "rtk", and UAV_MASS tuned to our drone). We wanted first to see how the RTK data are processed by the code. I though our connection from ReachM2 to Nuc was wrong so we put back ODOMETRY_TYPE = "gps" using the pixhawk gps to see if different but we had the same issue...  I put a screenshot of the 5_Control.log so you guys can see.  

2 : What are your settings in the correction input and position output tabs for the base (Reach RS2) and the rover(ReachM2) ? Personally I followed emlid documentation and I have precise position but I wonder if the Reach M2 is sending them correctly to the Nuc. (The M2 is connected to /dev/tty/ACM0 and we renamed it "rtk" following the procedure you mentioned in https://ctu-mrs.github.io/docs/hardware/px4_configuration.html.) I also put screenshots of my M2 and RS2 settings

Thank you again for your help,

Best regards

Frank 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

DanHert, 2021-03-19:
The odometry is shutting down because it expects data from a laser rangefinder (garmin lidar) for altitude estimation. Do you have this sensor on your drone, or no? If not, you can use the Z from RTK, but you have to be careful when flying over uneven terrain.

Regarding the RTK config, the rover should output NMEA messages over Serial. After you launch mrs_serilal rtk.launch, and read out the RTK messages. If you get L1_INT, your RTK is setup correctly and you have a RTK fix.

petrlmat, 2021-03-19:
Hi,
you need to add `garmin_down` to the `SENSORS` variable in `~/.bashrc` as the log suggests (i.e., `  export SENSORS="garmin_down"`. Of course assuming you have a downward garmin rangefinder to measure the height of the UAV. If not, you can alternatively change the altitude estimator in your odometry custom config to barometer only (i.e., `altitude_estimator: "HEIGHT"`), but I do not recommend this as the altitude measured by the PixHawk barometer is very unstable.

Regards,
Matej

kefran88, 2021-03-20:
Hello guys, 

First of all, thank you both for your (fast) answers. We have been able to observe the "L1_INT" as Dan mentioned. 

However, it seems the uav_state of the drone cannot be computed. I also have the feeling the RTK value is not shared in the whole system : we had a baudrate error message although we set it up at the same rate in the EMLID settings and in rtk.launch. I put you a screenshot of the 4_Control.log file and the Rtk.launch file.
In this picture you can observe this strange error message "Setup Pixhawk SD card"... We updated the firmware through QGC 
and the version of mrs_workspace is only 2 weeks old : what could be the issue here ? 

Thank you in advance for your help,

Best regards,

Frank

[image]
[image]
 [image]
[image]

DanHert, 2021-03-22:
Hello, the launch files in mrs_serial were a bit messed up. I fixed them ( `git pull` ), and it should work properly.
The odometry node is complaining that it is not getting data from pixhawk. Did you follow this guide - https://ctu-mrs.github.io/docs/hardware/px4_configuration.html ?
Mainly the part with sdcard config and ROS setup. 

Dan H.

kefran88, 2021-03-22:
Hello @DanHert , 

Thank you very much for your answer, 

Indeed, I had forgotten this step of the tutorial . Fixed now. We are moving slowly but surely towards our first flight, but we still have small issues,  mainly for the odometry : If you look at the status tab you can see that the odometry isn't working but on the Control.log you can read INFO messages saying the opposite. The only thing we don't have is a XYZ position of the drone but for me, it seems like everything is running so what could be the issue here ? (see screenshot)

Secondly, I updated the mrs serial after your commit but the problem with the rtk is still there. I am trying to look for the bug also. The ERROR message is everytime showing a new and wrong value of baudrate.  If you guys are able to fly that means I made a mistake somewhere in the EMLID configuration (but I doubt) or in the code (more possible)...

Thank you very much for your help,

Frank

[image]
[image]
[image]

DanHert, 2021-03-22:
The random baudrate value is just caused by a missing parameter. Did you really pulled mrs_serial? Make sure you are on this commit - https://github.com/ctu-mrs/mrs_serial/commit/531bf8cf261611fa035926ead9370b808f824409

Regarding the odometry, could you check the Odometry tab in tmux? It will report what is wrong.

Dan H.

ZakariaLaakel, 2021-03-23:
Hello,

I added some screens of what you requested. We're on the correct commit and from the odom tab it does not seem like there is data flowing, probably due to the baudrate error of the rtk.

[image]
[image]
[image]
[image]

Kind regards,

Zakaria

DanHert, 2021-03-23:
Ahh, sorry, I meant the Control tab, not Odometry (my bad, Odometry used to be separate from control, but now they are both in the same tab.) There should be some red errors there telling you what is wrong.

Is there still a problem with the mrs_serial node with the new launch files?

ZakariaLaakel, 2021-03-23:
I just tried to fly a few minutes ago. The baudrate error for the mrs_serial/launch/rtk.launch file is still present. I also added a screen of the Control tab.
[image]
[image]

DanHert, 2021-03-23:
Hmm, do you still have the regular GPS receiver on the drone? I see UTM: false, which would indicate that your drone does not have the regular GPS fix. The UAV actually needs the regular GPS as well, sicne it contains a magnetometer which is necesary for GPS flights (the one integrated inside pixhawk is not good enough). It also servers as a backup if RTK fails.

Regarding the mrs_serial, did you recompile it after pulling?

Dan H.

ZakariaLaakel, 2021-03-23:
If you mean the PixHawk GPS then we have it:
[image]

Yes I pulled and compiled the workspace containing mrs_serial, I had no errors when doing that.

ZakariaLaakel, 2021-03-24:
I tried to fly with the gps instead of the rtk for the state estimator and the barometer for the altitude estimator and got the following control tab:
[image]

DanHert, 2021-03-24:
Odometry is reporting that it could not load all non-optional parameters, so it probably threw an error earlier in the log.
Can you send the entire log instead of screenshots?

Dan H.

ZakariaLaakel, 2021-03-26:
Hello,

Here are 2 zip folders containing all the log files for 2 tests we did: 1 with the GPS as state estimator and BARO as the altitude estimator (GPSTest.zip) and the other has the RTK as state and altitude estimator (RTKTest.zip).

As you can see on the status tab, we got the GPS fixed for the GPSTest, but not with the RTKTest.
We tried to send the data from the RTK with another supported baudrate but it gave the same baudrate error. I also noticed there is an error with the checksum so the sent data is probably not even correct..
This can maybe explain why our X,Y and Z value of the RTK are saturating.

[GPSTest.zip](https://github.com/ctu-mrs/mrs_uav_system/files/6210946/GPSTest.zip)
[RTKTest.zip](https://github.com/ctu-mrs/mrs_uav_system/files/6210947/RTKTest.zip)
[image]
[image]

petrlmat, 2021-03-26:
The odometry node dies because it cannot load the parameter:
`Mär 24 16:54:20 [31m[ERROR] [1616601260.712423847]: [Odometry]: Could not load non-optional parameter /nuc2/odometry/state_estimators/fused_measurements/BARO[0m`
This is happening, because you put `BARO` as an active state estimator:
`Mär 24 16:54:20 [0m[ INFO] [1616601260.699088536]: [Odometry]: parameter 'state_estimators/active:	GPS, BARO[0m`
It is an altitude estimator, so it cannot be there. Your odometry config should look something like this:
```
# Lateral state estimator:
# OPTFLOW, GPS, OPTFLOWGPS, RTK, ICP, VIO, HECTOR
lateral_estimator: "GPS"

# Altitude state estimator:
# HEIGHT - rangefinder
altitude_estimator: "BARO"

# Heading state estimator:
# GYRO - gyro, COMPASS - gyro, compass, OPTFLOW - gyro, optflow, HECTOR - gyro, hector slam
heading_estimator: "PIXHAWK" 

# Active estimators are started at node launch and can be switched to during flight
state_estimators:
  active: ["GPS"]

heading_estimators:
  active: ["PIXHAWK"]
```

ZakariaLaakel, 2021-03-31:
Hello @DanHert and @petrlmat, 

Thank you for your answers, thanks to them the previous problems (except the weird baudrate one) have been solved.

After putting the origin at 0.0 0.0 in mrs_uav_general/config/world/world_simulation.yaml ,we observed now some serious X,Y,Z positions drift (10m order and even more) with both GPS and RTK as positions estimators. 
While we understand the X,Y position drift in GPS mode, we didn't expect that for the RTK case at all. The cm-precision is not visible, although in the EMLID tab , we have the RTK_FIX and the announced precision by the manufacturer.

I believe thus that the communication between the Reach M2 and the NUC is wrong. We observed "wrong checksum' along with the baudrate problem. To solve the latter, we tried to impose the baudrate to a constant value in the nmea_parser.cpp file but that didn't solve our issue. 

I put your the logs in zip file of the GPS flight and RTK flight, alongside 2 screenshots of the Status tab. 

Best regards,

Frank 
[image]
[image]

[RTK_flight_20210331.zip](https://github.com/ctu-mrs/mrs_uav_system/files/6238020/RTK_flight_20210331.zip)
[GPS_20210331.zip](https://github.com/ctu-mrs/mrs_uav_system/files/6238049/GPS_20210331.zip)

petrlmat, 2021-04-01:
Make sure, you have your UTM/LatLon origin set correctly for your location in your world file. `world_simulation.yaml` is for simulation, where the starting coordinates are different from your real coordinates. If the problem persists after fixing the origin coordinates, please record a rosbag. In this case the logs do not provide enough information for more in-depth debugging. It should suffice to remove the `waitForOffboard;` from the `Rosbag` tmux tab.

kefran88, 2021-04-02:
Hello petrlmat, 

Thank you for your answer, 

I changed world_simulation.yaml since it is the one mentioned in the ~/.bashrc file. I put my Long/Lat values in the utm_origin_lat_long as you can see it the figure. 

I saw that we can switch back and forth easily, but from your experience at MRS, which one do you advice me to use ? 

In the meanwhile, I will try a flight and take a look at the rosbag

best,

kefran88, 2021-04-02:
Hello @DanHert and @petrlmat 

We fixed our problem and now the RTK X and Y values are good (it takes some time for converging to the same value as in the ReachView app). 
The only problem is the RTK Z value. The Z variations seems to be precise but there is an offset that changes at each flight. We understand why you guys didn't advice to use the RTK for vertical position estimation. 

We still believe that we can achieve good results with this so our plan is to write a piece of code for this purpose. We would like to know how should we process the NMEA messages ? We already took a look at the NMEA_parser.cpp, most of the processing happens there but is there anything we should know in the mrs_serial ? 

If we achieve good results, we could put the code on the github for the community following your steps.

best regards,

Frank

closed 2021-04-09 by kefran88