ctu-mrs/mrs_uav_system #50: Switch controller between experiment and simulation

RaphaelNetels, 2021-03-25:
Hello MRS team,

For my project, I need to add an encoder. This means that the control is different in simulation and experiment. How do you switch between simulation and experiment in your controllers?

Best regards,

Raphaël

DanHert, 2021-03-25:
Hello, our controllers are exactly the same in simulation and in reality. Can you please expand you question a bit more? What do you mean by encoder?

When you fly with real drones, you need to change some of the .bashrc variables to correspond to your hw setup (and change the RUN_TYPE to uav). We also use tmux scripts instead of tmuxinator - https://github.com/ctu-mrs/uav_core/tree/master/tmux_scripts

Dan H.

RaphaelNetels, 2021-03-25:
Hello,

I will explain a bit more the project. I am developing a controller for two drones transporting a payload (beam) as can be seen in the image below. 

For the simulations, I subscribed in the controller to the topics in ROS that give the exact locations of the attachment points of the payload. For the experiments, I will use two rotary encoders placed underneath the UAVs that will give the angles and out of this, the position of the attachment points will be calculated. 

This means that I have the same inputs for my controller in the case of simulation and experiment (the attachment points of the load). So having two different controllers is therefore probably not the best solution and I was wondering how you switched. for example between real data from RTK and 'simulated' data from RTK.

I hope the question is a bit more clear.

Raphaël

[image]

DanHert, 2021-03-25:
I still probably do not understand you completely :D
To be clear and not to mix up the names, controller is just the part of SW which produces control actions based on a given
reference and the current state of the UAV.

The current state of the UAV is called odometry in our system (topics /$UAV_NAME/odometry/odom_main or /$UAV_NAME/odometry/uav_state) and it is independent of the controller. Our system is able to run multiple odometry estimators at the same time (e.g. RTK, normal GPS and vision) and switch between them during flight as needed or as requested. You just need to configure them correctly in the odometry config (https://github.com/ctu-mrs/mrs_uav_odometry/blob/master/config/state_estimators.yaml and other config files).

Is this what you need? To change the source of localization during flight? Or do you need to change the actual controller during flight? (that can also be done with the system).

Dan H.

closed 2021-04-05 by RaphaelNetels