ctu-mrs/mrs_uav_system #51: Same position goto command

ahmadipt, 2021-03-29:
Hi,
I have spawned 4 drones inside the world. Then send (0,0,2) command for all of them to see what will happen.
They collide with each other. it seems the MPC controller not working well in this situation.

petrlmat, 2021-04-15:
Hi, unfortunately we cannot help you with this issue without more information. The collision avoidance works by default, and you can test it by running the session `~/mrs_workspace/src/simulation/example_tmux_scripts/three_drones_gps`.

Without knowing how your setup deviates from the default one, it is impossible to assess the cause of the collision. Please provide your `session.yml` with tmux log and rosbag. Tmux logs should be by default recorded in `~/bag_files/` and to record a rosbag you can run `rosrun mrs_uav_general record.sh`.

ahmadipt, 2021-04-15:
Hi
Thanks for your reply.
I used the same script `~/mrs_workspace/src/simulation/example_tmux_scripts/three_drones_gps`
I just edited these command to go to the same position 

```
- 'export UAV_NAME=uav1;
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[10.0, 0.0, 2.0, 0.0\]\";
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[0.0, 0.0, 2.0, 0.0\]\"'
- 'export UAV_NAME=uav2;
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[-10.0, 0.0, 2.0, 0.0\]\";
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[0.0, 0.0, 2.0, 0.0\]\"'
- 'export UAV_NAME=uav3;
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[0.0, 10.0, 2.0, 0.0\]\";
  history -s rosservice call /$UAV_NAME/control_manager/goto \"goal: \[0.0, 0.0, 2.0, 0.0\]\"'
```

this is the only change i made to test that.

DanHert, 2021-04-15:
Hello, a few comments about collision avoidance:
The collision avoidance is just a safety mechanism for real-world experiments, it is not bulletproof and if you push it too much, it will lead to crashes. Your software, which is giving trajectories/references to the drones has to plan collision-free from the beginning, and not rely on our collision avoidance. The collision avoidance is there just for cases when something goes wrong,  and it should not be triggered normally.

In simulation, if you send all the drones to the same position, they end up above each other, and their laser rangefinders will actually not be pointed at the ground, but at the drone below, giving a bad altitude measurement, which will eventually lead you to a crash.

ahmadipt, 2021-04-15:
Thanks for the explanation.

closed 2021-04-15 by ahmadipt