ctu-mrs/mrs_uav_system #52: Errors Installing mrs_uav_system

O2KAN, 2021-04-05:
Hello, I am trying to install mrs_uav_system according to the given instructions. But I am taking these errors which are given in the attachment section. I tried to install it a couple of times but the errors didn't change. I am new to Linux, I don't know much about it. Could you help me to solve the errors? Thank you very much. Ps:  My computer has 16 GB RAM, Linux 18 installed. 

[image]
[image]
[image]

spurnvoj, 2021-04-15:
Hi, it seems that dependencies for the simulation were not properly installed. If you followed the instruction for mrs_uav_system, you should have the script for installing the dependencies in `~/git/simulation/installation/install.sh`. Try to run the script manually to see if it is without error. 
```bash
source ~/git/simulation/installation/install.sh
``` 

Note: Be sure that you have up-to-date version of the simulation repository. Call
```bash
cd ~/git/simulation
git pull
```

O2KAN, 2021-04-20:
Thank you for the reply. I will try it

throwawayacco, 2021-04-20:
I have a strong suspecion it might be related to a lack of memory. When I installed the system my installation failed due to a lack of memory when building the mrs_gazebo folder. The error you're experiencing seems to be an error caused from that.

Wizzya, 2021-04-27:
Hello i think i am facing the same issue but the catkin build does not work as intended. The image is attach below. Is there a work around for this?
/home/wiz/Pictures/Screenshot from 2021-04-26 23-33-11.png

Wizzya, 2021-04-27:
[image]
I think i might be facing a similar issue and it still doesn't work on my end too. Tried partition the storage space for more ram but to no avail.

spurnvoj, 2021-04-27:
Hi, please update `mrs_uav_system` repository and try to call install again. If it still crashes, modified these line https://github.com/ctu-mrs/mrs_uav_system/blob/master/scripts/set_mrs_workspace.sh#L45-L52 by adding parameter `-j $number_of_cores_for_build`. For example for the first line building using only 4 cores:
```bash
catkin build mavros -c --mem-limit 75% -j 4
```

throwawayacco, 2021-04-27:
My solution was to build it first with standard settings. 
Then removing the catkin configuration and running it again with -j1 -p1

O2KAN, 2021-04-27:
Hello, yesterday I figured out the problem. My logfile past the almost 2gb because of that, it had been giving the error. I deleted the logfile then installed again. Then it works now.

closed 2021-04-27 by O2KAN

klaxalk, 2021-04-27:
> Hi, please update `mrs_uav_system` repository and try to call install again. If it still crashes, modified these lines https://github.com/ctu-mrs/mrs_uav_system/blob/master/scripts/set_mrs_workspace.sh#L45-L52 by adding parameter `-j $number_of_cores_for_build`. For example for the first line building using only 4 cores:
> 
> ```shell
> catkin build mavros -c --mem-limit 75% -j 4
> ```

An additional option that I would recommend is to set up a swap of an appropriate size, such that your total RAM+SWAP is roughly 2.5 times the number of your CPU threads. Then, you won't have to be careful about that any more and the compilation will be way faster with all cores. Example for 16GB SWAP:

```bash
sudo swapoff -a
sudo dd if=/dev/zero of=/swapfile bs=1G count=16
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
grep SwapTotal /proc/meminfo
```