ctu-mrs/mrs_uav_system #53: Convert BACA to int in ROS

RaphaelNetels, 2021-04-06:
Hello,

As can be seen in the figure below, I am able to send an int with Arduino through the serial port and then subscribe to the topic using the BACA protocol. My question is the following, how do I go back from the payload to my integer? Is there a code somewhere that does this?

Thanks in advance for your response!

Raphael
[image]

DanHert, 2021-04-06:
Here is an example how to convert two 8 bit values into a 16 bit int:
```
int16_t range      = input_buffer[1] << 8;
range |= input_buffer[2];
```
Is this what you are looking for?

Dan H.

RaphaelNetels, 2021-04-06:
Yes, that was what I was looking for!

Thanks!

Raphael

closed 2021-04-06 by RaphaelNetels