ctu-mrs/mrs_uav_system #55: State-feedback between simulation and experiment

AlyDiane, 2021-04-14:
Hello,

For our project we implemented a load attached through cables to the UAVs. In the simulation mode, we subscribed to the gazebo topic "/gazebo/link_states" to get the position and the velocity of the corresponding link and implement it in our state-feedback. We would like to do the same in experiments by subscribing to the topics from ros serial package coming from our encoders.  For this purpose, the controller has to be able to make the difference between simulation and experiments in order to subscribe to the right topic.

Where/how can we change the mode to simulation or experiments such that our controller.cpp knows which topic to subscribe to for the state-feedback? Or do you see other ways that are better to program this functionality?

Thank you in advance,
Aly

DanHert, 2021-04-14:
Hello, we have a bash variable called RUN_TYPE, it is either 'simulation' or 'uav' (uav means a real world drone).
So you can check this variable in your lanuch file and remap the topics accordingly.

Dan H.

AlyDiane, 2021-04-14:
Hi DanHert, thank you very much for your response. It is now working!

Aly

closed 2021-04-14 by AlyDiane