ctu-mrs/mrs_uav_system #58: Take back the control using the RC transmitter during flight

kefran88, 2021-04-20:
Hi MRS team, 

I was wondering what is the procedure to take back the control with the RC transmitter during flight., mainly for manual landing when the drone is going craazy.

I am a bit scared to crash the drone since when it is taking off and during the whole operation, my thrust stick is down. If I take back the control (by switching off the offboard mode), I wonder if the drone will crash or if I put the thrust stick up (in order to put it back down for a manual landing), the drone will suddenly go way to high

Could you mention me the steps for safe switching between RC manual control and the offboard mode please ? 

Best regards, 

Frank

DanHert, 2021-04-21:
Once you switch off from the offboard mode, pixhawk will take over, so the position of the sticks on your RC is very important, if you have the throttle down, the UAV fill fly down. When we fly with GPS, we usually keep the Pixhawk in GPS mode and keep the throttle in the middle, so that when we switch out of offboard, the UAV will just stay hovering in place.

But we rarely have to do that, the system will handle a lot of problems on its own and safely land the drone for you.

kefran88, 2021-04-24:
Hi @DanHert 

Indeed, launching the Land through the Status tab is safer and easier. I just wanted to be aware of all possibilities to be able to ensure a safe flight. 

Thank you

closed 2021-04-24 by kefran88