ctu-mrs/mrs_uav_system #62: Flight autonomy of F450 drone

kefran88, 2021-04-26:
Hello MRS team, 

Me and my team have built 4 drone f450 following the configuration on the https://ctu-mrs.github.io/ and following this issue #15 .

We experienced a flight autonomy around 3 min, which is pretty low : The NUC is powered directly from 4S battery (just like @DanHert mentioned in issue #30 ), the same battery which provides voltage to power distribution board. During flights, whenever the battery is approaching the 12V, the drone lands and the NUC turns off a while after.

We wonder if you have also such short flight time, and if not what did you do to extent the autonomy of your 4S battery-powered drone ? 

Best regards ,

Frank

DanHert, 2021-04-26:
Hello, with a 4S 6750mAh battery, we get 10+ minutes of flight, dependant on the mass of the UAV.
Can you specify exactly what battery you use, what are the motors and propellers and what is the mass of the UAV?

And by the way, 4S battery should NEVER reach 12V, if it did, it is likely damaged. We set our lipo alarms to 3.5V/cell (14V for the whole pack), and when they go off, we land immediately.

Dan H.

kefran88, 2021-04-26:
Hello @DanHert ,

thank you for the fast reply 
 I forgot to mention that , to power the NUC, we soldered a DC plug at the DC input of the Holybro Power distribution board. This DC plug is connected to the NUC power input.

For a drone of 2.35 kg (everything taken into account) , we use : 

- battery : [Rhino 4S  6000 mAhbattery ](https://hobbyking.com/fr_fr/rhino-6000mah-4s-50c-lipo-battery-pack-w-xt90.html?___store=fr_fr)
- Motors and props : [ Ready Tosky RS2312-920 and DJI propellers ](https://ctu-mrs.github.io/docs/hardware/motor_tests.html) as mentioned on your website

Has the alert been setup in QGC or in the [px4_config.yaml](https://github.com/ctu-mrs/mrs_uav_general/blob/6048ed20672573cc4102ccef19d07082c6f4d457/config/px4_config.yaml) ?

Best,

Frank

DanHert, 2021-04-26:
Hello, the alarm is an external one, like this:
[image]
It is plugged to the balance connector and it will beep loudly when the alarm voltage is reached.

Did you charge the battery fully? (to 16.8V). Is the battery new, or old? 2.35 kg is on the limit of the payload capability of this configuration, but you should still get more than 3 mins.

kefran88, 2021-04-26:
Hello,

I am confused : isn't the max voltage for a 4S battery 14.8V ? As it is mentioned on the datasheet ? 
When we charge, we go up to 14.8V-15V sometimes, but it won't go higher. 

The batteries are 6 months old.

DanHert, 2021-04-26:
14.8V is the nominal voltage (3.7 V per cell), but at this voltage the cells are less than 50% charged. You need to charge them to 4.2 V per cell for flight, that is fully charged. Learn more here - https://learn.adafruit.com/li-ion-and-lipoly-batteries/voltages

Dan H.

kefran88, 2021-04-26:
Well well,  

We made a misunderstanding, we will correct that. Thank you for your answers !

kefran88, 2021-04-26:
Hello again @DanHert ,

I forgot to ask but to make the drone lands automatically when the battery voltage goes below 14V or whatever the threshold, what are the parameters and config files that I need to change ? (like `min_voltage` in
[mrs_uav_general/config/px4_config](https://github.com/ctu-mrs/mrs_uav_general/blob/6048ed20672573cc4102ccef19d07082c6f4d457/config/px4_config.yaml)  ?

DanHert, 2021-04-27:
Hello, you can do this with Pixhawk settings, but you need to properly calibrate its voltage measurement correctly.
But I am not sure if this feature will work properly during autonomous offboard flight (we never tested that). We do not use this feature, we simply monitor the flight time and the lipo alarm, and land the drone when the battery is low. 

The system will also land the drone automatically when the thrust needed for flight is over 80 percent of the max thrust, which will happen when your battery is dying. But you should never reach this far, as the battery will be damaged.

The battery voltage is also sent from Mavros to the onboard computer, so you can easily program a node that will monitor it and land the drone when the voltage is at a  specified level.

Dan H.

closed 2021-04-27 by kefran88