ctu-mrs/mrs_uav_system #64: Need help identifying files for drone coordinates

Wizzya, 2021-05-05:
Hello mrs-team,
I am currently having trouble with identifying the files that are required and used to change the x,y,z coordinates of the drone spawn. I am mainly interested in changing the z value of the drones. This is to help me with another project that requires me to navigate through a series of varying ground terrains- eg. mountains. I also need some guidance on how to integrate the world files coordinate to respond to the changing altitudes of the ground since the default world file configurations and mrs_launch files interact independently from one another.

spurnvoj, 2021-05-05:
Hi, you can specify the initial position of drone in simulation using `--pos_file` parameter. See this [session.yml](https://github.com/ctu-mrs/simulation/blob/ceb42caee1cef876d7d3cba413ca700539d0ebed/example_tmux_scripts/one_drone_gps_custom_configs/session.yml#L21) as an example. The initial position can be loaded from csv or yaml file (see [pos.yaml](https://github.com/ctu-mrs/simulation/blob/ceb42caee1cef876d7d3cba413ca700539d0ebed/example_tmux_scripts/one_drone_gps_custom_configs/pos.yaml)).

For changing of GPS coordinate origin for our system, you have to update this [file](https://github.com/ctu-mrs/mrs_uav_general/blob/dfd4b596a983ef899473890b37748f74e217df32/config/worlds/world_camp.yaml#L8-L9) to be in match with origin of your world. Or do it in opposite way and modify origin in your world file to match the one in the our setting. See https://github.com/ctu-mrs/mrs_uav_system/issues/22

Wizzya, 2021-05-10:
How am i able to launch the custom configs into the session.yml files of the tmux examples scripts. For example, i want to spawn the one_drone_gps with the uneven_plane and then add some obstacles inside as shown in the world.simulation.yaml.
So when i set waypoints in this simulation, i noticed that there is a hdg value that reads the height or the terrain with respect to the ground. Will appreciate if i could get guidance to where is this file located.

Wizzya, 2021-05-10:
Update: tried inputing this in one_drone_gps by replacing the gazebo world part with:
 - gazebo:
      layout: tiled
      panes:
        - roslaunch mrs_simulation simulation.launch gui:=true world_file:='$(find custom_gazebo_resources)/worlds/custom_world.world'
        - waitForRos; roslaunch mrs_simulation simulation.launch world_name:=world_simulation.yaml gui:=true
        - waitForOdometry; gz camera -c gzclient_camera -f uav1; history -s gz camera -c gzclient_camera -f uav1

But only boots the one drone gps into an empty world
not sure what i did wrong to prevent the config world from spawnning

spurnvoj, 2021-05-10:
Hi, have you seen this https://ctu-mrs.github.io/docs/simulation/custom_world.html? In the example above you are running two times the gazebo simulation with world. Usually, the empty world in the gazebo means that the world file doesn't exist (the path to the world file is not correct) or contains some error.

spurnvoj, 2021-05-17:
@Wizzya  Have you managed to run it as you wanted?

Wizzya, 2021-05-18:
I can run default simulations thanks for your help. But cuurrently still having trouble with launching custom worlds

closed 2021-05-18 by Wizzya