ctu-mrs/mrs_uav_system #66: FTDI board

PhilPhal, 2021-05-05:
Hi MRS team,

During the construction of a second f450 drone, I come across some problems with the communication between the Nuc and the PixHawk. The "Rx" blue light of the FTDI board doesn't light up when I connect it. I disassembled my first f450 drone to make sure I did the same and I can't find any differences. I also tried placing the FTDI board and cables on the first drone, and there the connection succeeds.

Do you have an idea what could be the issue?

Thank you in advance.
Philippe

spurnvoj, 2021-05-05:
Hi, check the pixhawk sd card - https://ctu-mrs.github.io/docs/hardware/px4_configuration.html#sd-card-setup.

PhilPhal, 2021-05-06:
Hi,

Thank you for the quick response. I was unaware of that setup... It fixed the problem!

Have a nice day,
Philippe

closed 2021-05-06 by PhilPhal