ctu-mrs/mrs_uav_system #77: RTK settings

RaphaelNetels, 2021-05-13:
Hello MRS,

I am still struggling to use the RTK correctly for the real hardware. I want to increase the rate at which position data is sent to the controller (from 1Hz to 10 Hz) but I am not sure what to change. I tried changing the GNSS update rate to 10 Hz but it does not seem to change anything. In the figures, the settings of the base (reach emlid RS2) and the rover (reach emlid M2) can be seen. Is there something I did not correctly or that can be improved?

Base settings:
[image]
[image]
[image]
[image]

Rover settings:
[image]
[image]
[image]
[image]

Thank you for your response and time!

Best
Raphaël

DanHert, 2021-05-13:
[image]

Change this to 10 Hz

Dan H.

RaphaelNetels, 2021-05-13:
Thank you!

Raphaël

closed 2021-05-13 by RaphaelNetels