ctu-mrs/mrs_uav_system #78: Questions concerning the usage of UAV_mass & motor_params in simulations

ZakariaLaakel, 2021-05-17:
Hello MRS team,
 
I had a few questions concerning the usage of your software during simulations:
 
First of all we noticed that the parameter in the bashrc 'UAV_MASS' is not working for the simulations. Indeed if the RUN_TYPE is set to 'uav' then the mass used during experiments is the one specified in the bash, but if it is set to 'simulations' it seems like this parameter is totally ignored. 
The only way we found to change successfully the mass during simulations is by modifying the default mass in the core.launch file. However we're not sure this new mass is correctly loaded everywhere and we were asking if there isn't another (better) way to change the mass and use it everywhere for simulations. Is it possible for example to define it as a parameter in custom config files? So that it would allow us to stil use the F450, but just overwrite the mass and without changing your code in the mrs_workspace. 
 
The second question is a bit similar to the first one but related to the motor parameters of our UAV.
At the moment we copied the different motor_parameters.yaml files inside our own workspace but we are unable to use them in simulations or real experiments because these parameters are called via 
 
<arg name="motor_params_file" default="$(find mrs_uav_controllers)/config/$(arg RUN_TYPE)/$(arg UAV_TYPE)/motor_params_$(arg PROPULSION_TYPE).yaml" />
 
We need more something like
 
<arg name="motor_params_file" default="$(find our_controllers_package)/config/$(arg RUN_TYPE)/$(arg UAV_TYPE)/motor_params_$(arg PROPULSION_TYPE).yaml" />
 
But we do not want to change anything in the mrs_workspace, neither we want to break the whole structure since we want to stay up to date with your software. So changing this line of code the whole time is not really
convenient for us. Is there a way to set these motor parameters of the core.launch file such that the line above is changed from "find mrs_uav_controllers" to "find our_controllers_package"?
 
The last question is still related to the motor parameters. Our real drone is a F450 with 'Readytosky 2312 920KV motors' and 'Aliexpress plastic 9450 self-tightening' propellers. 
However we think we do not use the same ESCs as you, we use the 'Simonk 30A' ESCs. Our drone is heavier than your default mass for the F450 (2.35 kg vs 2.0 kg).
 
We see [here](https://github.com/ctu-mrs/uav_core/tree/master/miscellaneous/thrust_constants) that the parameters are calculated in 2 different ways (the 2 folders).

Which script should we use? The one from 'uav_thrust_curve_estimation'? In that case we should use the parameters that are returned by the script 'f450_readytosky.m' , right?
 
Do these parameters change because we have other ESCs (Simonk 30A) and mass?
 
Thank you for your answers.
 
Zakaria

klaxalk, 2021-05-18:
> Hello MRS team,

Hey, Zakaria. Thanks for posting this question. As you already know, you just found a spot in our system, which is not well prepared for user modifications. But I hope we will come up with a way that will make it possible. Let me go through your post.

> I had a few questions concerning the usage of your software during simulations:
>
> First of all we noticed that the parameter in the bashrc 'UAV_MASS' is not working for the simulations. Indeed if the [...]

You are correct. The .bashrc parameters are intended for configuring a real UAV. On the other hand, relying on the .bashrc for configuring simulation sessions would be cumbersome, since you can have multiple sessions at hand with different UAV configurations. Therefore, some parameters (the ones differing from their default values) are overridden in the tmuxinator sessions, e.g., the `one_drone_gps` example session has this at the top:
```
pre_window: export UAV_NAME=uav1; export RUN_TYPE=simulation; export UAV_TYPE=t650; export WORLD_NAME=simulation; export SENSORS="garmin_down"
```
The mass parameter is a special one since it is ignored completely and instead, it is "hardcoded" in our launch files. Originally, my idea was to make it easier for our user, therefore, when switching the UAV_TYPE, the user should not need to insert the nominal mass as well.

> The only way we found to change successfully the mass during simulations is by modifying the default mass in the core.launch file. However we're not sure this new mass is correctly loaded everywhere [...]

Ye, I think it will be better to store the masses in a config file, which could be also customized. I will make that happen.

> The second question is a bit similar to the first one but related to the motor parameters of our UAV.
> At the moment we copied the different motor_parameters.yaml files inside our own workspace but [...]

There are two ways of solving this.
1. the quick and dirty way is to put your motor params as a part of custom configs for the `control_manager`, `uav_manager`. The motor params config is ultimately loaded into these two nodes, so this method will work.
2. I will make an alternative way of defining a `custom_motor_params_file` argument for the `core.launch`.  

> But we do not want to change anything in the mrs_workspace, neither we want to break the whole structure since we want to stay up to date with your software. So changing this line of code the whole [...]

See the previous comment. I will let you know.

> The last question is still related to the motor parameters. Our real drone is a F450 with 'Readytosky 2312 920KV motors' and 'Aliexpress plastic 9450 self-tightening' propellers.
> However we do not use the same ESCs as you, we use the 'Simonk 30A' ESCs. Our drone is heavier than your default mass for the F450 (2.35 kg vs 2.0 kg).
> 
> We see [here](https://github.com/ctu-mrs/uav_core/tree/master/miscellaneous/thrust_constants) that the parameters are calculated in 2 different ways (the 2 folders).
> 
> Which script should we use? The one from 'uav_thrust_curve_estimation'? In that case we should use the parameters that are returned by the script 'f450_readytosky.m' , right?

The subfolder `motor_thrust_curve_estimation` contains scripts for calculating parameters for a single motor+propeller on a test stand. The subfolder `uav_thrust_curve_estimation` contains scripts for calculating parameters for a whole UAV by inputting 2 or more (mass, thrust) points obtained by flying. The scripts are very similar, just the UAV script has the number of motors as a paramater. All the scripts output the  A and B parameters for the config file.

> Do these parameters change because we have other ESCs (Simonk 30A) and mass?

Yes, it is possible that due to having different ESCs, the parameters will be different. But not because of the mass. When the UAV mass is different, just change the UAV_MASS variable and the UAV will cope with that. And even without that, the system will estimate the mass difference (but it needs good motor params) and it will fly just fine.

> Thank you for your answers.
> 
> Zakaria

Thanks for asking. Please, leave this issue open. I will update the system and write a documentation page about it. I will keep you posted in here.

RaphaelNetels, 2021-06-16:
Hello

I was trying to do something similar with the masses of the drone and was wondering if there was any update on this issue?

Thank you for your response!
Raphaël

klaxalk, 2021-06-17:
Work in progress, I will lock it this evening.