ctu-mrs/mrs_uav_system #83: Multiple drone test

PhilPhal, 2021-06-23:
Hi MRS,

I would like to perform a real-life test with multiple drones but can't find a template of a just_flying.sh to do this. Is there one somewhere? I could use an example ;)

Thank you in advance.

Best regards,
Philippe

lidiaxp, 2021-06-28:
Hi, the MRS's explanation to use multiple drones are in https://ctu-mrs.github.io/docs/simulation/drone_spawner.html

I use the files in [this paste](https://drive.google.com/drive/folders/1JeDfhyCfbOXd9G_YE1Xitkr4HScl4GOp?usp=sharing) to launch multiple UAVs. You just need the 13 line on session.yml to set your launch. I will put the launch as sample too, the main change is in the line 14 to set the name of your world, and in the line 17 to set the path of your world (both in the launch file).

DanHert, 2021-06-28:
@lidiaxp I think Phil wants to fly with real drones, not in simulation .

PhilPhal, 2021-06-28:
Yes indeed! Thank you very much for the help, I will take a look at it.

Best regards,
Philippe