ctu-mrs/mrs_uav_system #85: Change the velocity🚀

lidiaxp, 2021-06-28:
Is it possible to change the UAV speed on this system? I'm using the F-450. By the calculations I made the default speed of the UAV is 1.6 m/s, but I would like to know if there is any parameter to increase or decrease this speed. Or a way to turn off control for the UAV to respond to cmd_vel.

Thanks in advance

klaxalk, 2021-08-06:
The UAV speed is defined by the system "constraints", which are managed by the [ConstraintManager](https://github.com/ctu-mrs/mrs_uav_managers#constraintmanager). The speed defined within the constraints is the "maximum limit", so the trackers should not allow you to fly faster. So those are the "limits".

However, if you want to control the UAV by specifying the desired speed, please, use the **control_manager/velocity_reference** topic while the **MpcTracker** is active.

lidiaxp, 2021-08-11:
I'm trying to set the speed by constrainst_manager.yaml setting the speed and acceleration to 4, and the jerk and snap to 50. But the maximum speed I get is 1.4, do you know why?

MpcTracker is active by default, and even if you set it manually, the control_manager/velocity_reference topic doesn't exist.

klaxalk, 2021-09-09:
Please check the topic `control_manager/current_constraints` to see what is currently set. Beware, If you fly with the MpcController, the current constraints are overridden to much slower than what you set with the ConstraintManager. Also, the topic `control_manager/velocity_reference` should exist since 4108be5 in uav_core.

lidiaxp, 2021-12-01:
I am using MpcTracker and MpcController to fly. When I check `control_manager/current_constraints` the constraints are fast (horizontal speed = 8) but when the UAV takeoff the constraints switch to slow (horizontal speed = 1). I updated the MRS and used the topic `control_manager/velocity_reference` but nothing happens (have tried with services too).

```
rostopic pub /uav1/control_manager/velocity_reference mrs_msgs/VelocityReferenceStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
reference:
  velocity: {x: 5.0, y: 5.0, z: 5.0}
  altitude: 0.0
  heading: 0.0
  heading_rate: 0.0
  use_altitude: false
  use_heading: false
  use_heading_rate: false" 
```
I am using RTK as odometry and setting the constraints to fast as follow [in these config files (.yaml)](https://drive.google.com/drive/folders/1PESJcv_xZUOnbeBtHJFNIXPvKu-H3sWQ?usp=sharing). And calling them in the start.sh `'Control' "waitForOdometry; roslaunch mrs_uav_general core.launch WORLD_FILE:=./world.yaml config_constraint_manager:=./custom_configs/constraint_manager.yaml config_odometry:=./custom_configs/odometry.yaml"`.

How do I keep the fast constraints after the takeoff?