ctu-mrs/mrs_uav_system #95: one_drone_gps_baro drone's altitude keeps increasing🐞

bconvens, 2021-07-23:
Hi, I just installed a new version of the mrs_uav_system on Ubuntu 18. I noticed the default simulation script [one_drone_gps_baro](https://github.com/ctu-mrs/simulation/tree/master/example_tmux_scripts/one_drone_gps_baro) is not working. The drone's altitude keeps increasing forever.

I would like to run a simulation and a hardware experiment for which no rangefinder is required for altitude estimation. Moreover in the swarm case I don't want the rangefinder to switch height when another uav suddenly flies below it.

Thanks in advance for solving the issue.
Bryan

petrlmat, 2021-07-24:
Hi, this is a bug in a recently added feature, which should prevent the UAV to collide with an obstacle or ground below it. Until we fix it, please disable the feature by adding these lines into your `uav_manager.yaml` custom config file:
```
min_height_checking:
  enabled: false
```

petrlmat, 2021-07-24:
Fixed in https://github.com/ctu-mrs/mrs_uav_odometry/commit/d65c7d4c6c7b80f50d6d467f55d05f5197847686

closed 2021-07-24 by petrlmat