ctu-mrs/mrs_uav_system #110: Make the swap persistent after reboot.

spurnvoj, 2021-09-09:
Current [install](https://github.com/ctu-mrs/mrs_uav_system/blob/master/install.sh#L48-L53) script shows steps how to create swap and use it. However, the swap is not persistent and will not be used after restarting pc. We need to modify `/etc/fstab` file for it. (see https://linuxize.com/post/how-to-add-swap-space-on-ubuntu-18-04/).

Note: Some people already have a swap file. Therefore, we should increase it instead of creating a new one.

klaxalk, 2021-09-09:
We did not have to do this on any of our drones. Are you sure you are doing this correctly? I am using:
```bash
sudo swapoff -a
sudo dd if=/dev/zero of=/swapfile bs=1G count=16
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```

spurnvoj, 2021-09-10:
Yes, I am using the same commands (see install script). To have swap permanent, you have to modify the `fstab` file. Here is another [link](https://itsfoss.com/create-swap-file-linux/) that mentions that too. They also show how to make swap larger if you already have one.

klaxalk, 2021-09-10:
Ye, maybe, if you select to have "no swap" during Ubuntu installation. We should have the swap-setting script universal, therefore, let's also update https://github.com/ctu-mrs/uav_core/blob/master/miscellaneous/scripts/set_swap.sh.