ctu-mrs/mrs_uav_system #117: Allow (again) to see and set non-ctu controllers and tracker in mrs_uav_status window🚀

bconvens, 2021-11-24:
Hi,
Would it be possible to uncomment the two commented else statements [in this commit](https://github.com/ctu-mrs/mrs_uav_status/commit/af71a0c73b7f43618b81b4b31a2a08bbd81adf1d)?
This would allow again to display and set the non-ctu controllers and trackers which were created by other users.
Thanks!
Best, Bryan

klaxalk, 2021-11-24:
Hey Bryan, you can make you controller and tracker visible by loading them with
```yaml
human_switchable: true
```, as in [here](https://github.com/ctu-mrs/mrs_uav_managers/blob/8d55923fd2f971b6b4f5da7340fbb7930268acef/config/default/controllers.yaml#L7). We hide all the controllers and trackers by default to prevent accidental switching into something that is not supposed to be easily activated.

Sincerely, Tomas

matemat13, 2021-12-07:
I'm closing this issue as it is not a bug, but a feature ;)

closed 2021-12-07 by matemat13