ctu-mrs/mrs_uav_system #118: Toubles for running Rviz after installing the mrs_uav_system🐞

Raphaelb10, 2021-11-27:
After installing the installing mrs_uav_system software, I get this error : 
/opt/ros/melodic/lib/rviz/rviz: symbol lookup error: /opt/ros/melodic/lib/libtf.so: undefined symbol:_ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEb 
after trying to run : rosrun rviz rvis (with roscore already running in an other tab.)

This can also be reproduce from ctu's code as well by launching the simulation for the one_drone_gps : 
 cd ~/git/simulation/example_tmux_scripts/one_drone_gps
./start.sh
The tab 7 will display the same error.

Several things have been tryied to solve this issue : 
-updating (sudo apt-get update and  sudo apt-get upgrade)
-Reinstalling Rviz (following : [This](https://www.howtoinstall.me/ubuntu/18-04/rviz/)
-Found [this](https://github.com/stereolabs/zed-ros-wrapper/issues/663#issuecomment-761614362) issue, about a missing libtf package dependency. I installed all (e.g. sudo apt install ros-melodic-tf) but did not help either.

This has been tested on 2 separated PC with same results. 
Thx in advance.

klaxalk, 2021-11-29:
Hey, thanks for contacting us. I have not experienced this issue, yet. So we will need to do some detective work.

The problem could be related to the fact, that we build our own tf2 fork within the uav_core. However, I see that your program loads tf2 from /opt, which suggests that the sourcing order, or the extending of workspaces could be somehow broken.

Could you, please, provide information on
- how / what you source (.bashrc),
- what is your workspaces structure and what are your workspaces extending?

Sincerely, Tomas

Raphaelb10, 2021-12-01:
Hi Tomas,
Thank you for the fast response.
Here is joined my .bashrc file as you requested. (I've put it in a .txt file because I could not import hidden files here) :

[bashrc.txt](https://github.com/ctu-mrs/mrs_uav_system/files/7633333/bashrc.txt)

Sorry but I am not sure to understand your second question, can you give more precision's please ? 

Thank you again.

Kind regards,
Raphaël

matemat13, 2021-12-06:
Hello Raphaël, the symbol you are looking for should be located in `/opt/ros/noetic/libtf2_ros.so`. You can verify this by running the command
```
nm -gD libtf2_ros.so | grep _ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEb
```
 * If the symbol is there, then the `libtf2_ros.so` library is not loaded into Rviz for some weird reason.
 * If the symbol is not there, then your ROS installation is probably broken... or maybe if it's there but has a different signature (i.e. the argument list is different), you have the `libtf2_ros.so` library from a different ROS version than the `libtf.so` that attempts to use it.
 * If the library is not there at all (`nm` will complain that there is `No such file`), install it.

klaxalk, 2021-12-07:
Sorry for not replying earlier. Thanks @matemat13 for the reply. Let's continue from there. I see that the .bashrc contains multiple sources of different workspaces. It is possible that some of them could be poorly set up.

bconvens, 2021-12-15:
Hello, thanks for the replies. I got the same error when reinstalling mrs_uav_system on ubutnu 18. 
I followed [these suggestions](https://github.com/ctu-mrs/mrs_uav_system/issues/118#issuecomment-986962820).
I got this
[image]
So I tried to install it, but I get in a loop of to be installed packages, which I fail to install properly. I think the same issue as [this](https://answers.ros.org/question/365657/error-while-loading-shared-libraries-libtf2_rosso/).
[image]

Finally I noticed that when I comment both of the following lines in the .bashrc, `rosrun rviz rviz` works again.
`source /home/bryan/mrs_workspace/devel/setup.bash`
`source /home/bryan/workspace/devel/setup.bash`

Any further suggestions?

Thanks! ;-)

matemat13, 2021-12-15:
Sorry, I didn't notice that you are still on Melodic, then the library containing the symbol's implementation may be located elsewhere.

It seems that the problem is actually related to what Tomáš suggested - you're sourcing multiple workspaces in your `~/.bashrc`, which creates some conflicts. In general, you should always only source **one** workspace (the one located in `~/workspace` in your case, most probably). See our [workspace tutorial](https://ctu-mrs.github.io/docs/software/catkin/managing_workspaces/managing_workspaces.html).

klaxalk, 2021-12-16:
Oh, that is strange. This is some "soft" incompatibility then. It is not like we are not testing against Melodic (https://github.com/ctu-mrs/mrs_uav_system/runs/4541480152?check_suite_focus=true#step:4:5078). But I am not sure if someone ever tested Rviz on Melodic in the past year...

klaxalk, 2021-12-16:
I will "blacklist" our tf2 automatically for Melodic then...

maxhofidrone, 2021-12-17:
> Sorry, I didn't notice that you are still on Melodic, then the library containing the symbol's implementation is located elsewhere and the package to install it called otherwise. In any case, the library most certainly **isn't** located in your home folder. You must of course call the `nm` command from the folder where it **is** located. This is `/opt/ros/noetic` in my case, but if you're on Melodic, then it will be elsewhere (I'd start looking in `/opt/ros/melodic`).
> 
> However, it seems that the problem may actually be related to what Tomáš suggested - you're sourcing multiple workspaces in your `~/.bashrc`, which creates some conflicts. In general, you should always only source **one** workspace (the one located in `~/workspace` in your case, most probably). See our [workspace tutorial](https://ctu-mrs.github.io/docs/software/catkin/managing_workspaces/managing_workspaces.html).
> 
> Actually now that I think about it, the error may be on our side. @klaxalk we are building `tf2_ros` on our own, right? But I've checked and as far as I can tell, we're building it from the `warning_fix` branch, which is one commit after `noetic-devel`, which is not backwards-compatible with ROS Melodic! So this is an actual bug in our system, if we want to claim ROS Melodic backwards-compatibility, and we should fix it :)
> 
> @bconvens, @Raphaelb10 - for you, the fix should be the following:
> 
> ```
> roscd tf2_ros;
> git checkout melodic-devel;
> catkin build;
> ```
> 
> then clean and re-build your all other workspaces you may be using (such as the `~/modules_workspace` or `~/workspace`). This should hopefully be a hot-fix for this bug.

Thanks that fixed for me the same bug!

bconvens, 2021-12-20:
Thanks a lot @maxhofidrone, this solves the issue on Melodic!

matemat13, 2021-12-20:
> Thanks a lot @maxhofidrone, this solves the issue on Melodic!

I'm glad it worked, although I'm not sure why you're thanking Max :D

@klaxalk did you do the blacklist so that we can close this issue with a proper fix?

klaxalk, 2021-12-20:
@matemat13 not yet, will do today.

klaxalk, 2021-12-21:
The custom tf2 is being blacklisted when using automated installation in the MRS UAV System. For the sole users of the uav_core, I added more info in the [README](https://github.com/ctu-mrs/uav_core/blob/master/README.md).

closed 2021-12-21 by klaxalk