ctu-mrs/mrs_uav_system #140: gzserver: symbol lookup error

O2KAN, 2022-04-16:
gzserver: symbol lookup error: /home/ozkan/mrs_workspace/src/simulation/ros_packages/px4_firmware/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so: undefined symbol: _ZN8mav_msgs4msgs17CommandMotorSpeedC1Ev

I am facing this error, could you help me get through this error?

klaxalk, 2022-04-19:
Hey, I would need much more information to help you. Are you building an up-to-date version of the system? Are you sure you are not checking out some custom (meaning your's) version of the px4 package? The build passes on our servers. If you just need to use our system, I recommend using [singularity](https://github.com/ctu-mrs/mrs_singularity/) images.

O2KAN, 2022-04-21:
Hei, Yes I am using up to date version of the system. I am using px4 when comes with your system not to install px4 firmware again. My problem is fixed somehow while trying to install jmsim simulation environment. Now working fine, thank you.

closed 2022-04-21 by O2KAN