ctu-mrs/mrs_uav_system #141: ROS2 version

sante, 2022-05-11:
Do you have plans to make a version compatible with ROS2?

klaxalk, 2022-05-16:
Hey, yes, possibly in 2023.

penickar, 2023-01-20:
So where is it? :-D

klaxalk, 2023-01-20:
I will start preparing for the transition soon. If there are no hard limitations, we could be running ROS2 in autumn.

yizh7963, 2023-09-27:
Could you please let us know if any progress of MRS_UAV_System compatible ROS2? Thanks

klaxalk, 2023-10-25:
Hey, no. We are postponing the transition to 2024. ROS2 is still too _finicky_ to support our system. There are too much realtime execution-related problems that would not allow us to conduct high-rate feedback control onboard the drones.

kursatkomurcu, 2024-05-01:
Will you release ROS2 version?

klaxalk, 2024-05-02:
Hey, we will finish this season using the ROS1 and, hopefully, the autumn and winter months will allow us to transition to ROS2.

Thxssio, 2024-08-03:
Hello, how are you? Any news about ros2 compatibility?

Mechazo11, 2024-08-21:
Hi @klaxalk 
This research work seems very promising specially given the group is working on both physics-based simulation and real-world deployment. I have had some practice in moving ROS 1 packages to ROS 2. Any way I can be of any help in transitioning the system to ROS 2?

klaxalk, 2024-09-09:
We have a long-term running small project (https://github.com/ctu-mrs/ros2_examples) where we want want to first test in implement all the features of ROS that we use before we start the migration. Every year, we check what the state is. You could maybe check our comments in readme. If you see something for which you know the answer, pls PR :-).

Otherwise: I am dreading when we finally start testing the different DDSs. What is the current state of these things? Is there a DDS that would allow for fast transmission and discovery while allowing to transmit large chunks of data such as images and pointclouds?

klaxalk, 2025-06-22:
Hey, I have a positive update for you. We are getting close to having the core of the system ready in ROS2. The rest will follow shortly. You can follow the progress here: https://ctu-mrs.github.io/ros2_obsidian_knowledgebase/main_canvas.html

closed 2025-09-19 by klaxalk