ctu-mrs/mrs_uav_system #150: 📚 Backward incompatibility: Updates for px4 firmware v1.13.2

spurnvoj, 2023-01-17:
This is an important notice for everyone using our system. In the case the MRS system was/is already locally built on your system, you need to do the following steps to have the system properly working:

1. Remove uav_core and simulation and clone them again. 
2. Remove all folders `sitl_uav$` in `~/.ros` folder.
3. Call installation script in the `uav_core` repository.

nrlhozkan, 2023-02-20:
Hello. When I do that on my computer, files inside simulation/ ros_packages do not have folders like px4_firmware and vs. any longer. But, Inside git, the probably old folders, are still there.  What should İ do?

spurnvoj, 2023-02-21:
@nrlhozkan did you follow the first point? 
> Remove `uav_core` and `simulation` repositories and clone them again. 

If yes, have you called `gitman install` in the root of the repositories? We are using `gitman` for managing the git repositories https://ctu-mrs.github.io/docs/software/gitman.html.

nrlhozkan, 2023-02-21:
I have one workspace and one git folder. I delete, uav_core and simulation in src in my workspace. Then clone again in src. Then I did catkin clean to clean the workspace. Later delete sitl_uav in .ros. After that call the installation.sh in uav_core/installation. Finally build my workspace with catkin build. I did not  use gitman install.

spurnvoj, 2023-02-21:
If you installed our system using the native way as it is described in README here https://github.com/ctu-mrs/mrs_uav_system#native-installation-not-recommended, the folders in workspace/src should be symlinks to the git folder. Therefore, you can delete `simulation` and `uav_core` folders in the workspace and the same files in the git folder. Then, go into `mrs_uav_system` folder in `~/git`, update it, and call the install script:
```bash
cd ~/git/mrs_uav_system
git checkout master
git pull
./install.sh -g ~/git
```
This should properly download and compile everything.

klaxalk, 2023-02-21:
And if you cloned `uav_core` or `simulation` manually, then you **need** to run `gitman install` to download the submodules. That will populate the respective `uav_core/ros_packages` and `simulation/ros_packages` folders.

matemat13, 2023-03-01:
If the `install.sh` script in `uav_core` fails with an error
``` Python
ImportError: cannot import name 'OS_RASPBIAN' from 'rospkg.os_detect' (/usr/local/lib/python3.8/dist-packages/rospkg/os_detect.py)
```
see issue #154 for a solution.

WingRS, 2023-03-15:
When doing the steps mentioned above, I faced an issue the install.sh script doesn't do `sudo apt update` before calling for installation. This is a simple issue but might confuse some users

WingRS, 2023-03-15:
When calling the install script in uav_core, I have faced an issue with some un-updated IP addresses, and maybe we could add `sudo apt update` in the install script. This is a simple issue but might confuse some people

closed 2025-09-19 by klaxalk