ctu-mrs/mrs_uav_system #177: URGENT: How to completely remove mrs system full?

arjungupta72, 2024-02-06:
I had installed mrs uav system using 
`sudo apt install ros-noetic-mrs-uav-system-full`.
I want to completely remove it from my sytem. I tried using the command
`sudo apt remove ros-noetic-mrs-uav-system-full` and `sudo apt purge ros-noetic-mrs-uav-system-full
` but it only remove some files and the ros noetic distribution folder still contains folders and files related to mrs. Please help.
MY system config:
ROS Noetic
Ubuntu 20.04.6 LTS

klaxalk, 2024-02-06:
https://github.com/ctu-mrs/ppa-stable?tab=readme-ov-file#uninstalling-packages-from-this-ppa

arjungupta72, 2024-02-06:
> https://github.com/ctu-mrs/ppa-stable?tab=readme-ov-file#uninstalling-packages-from-this-ppa

Thank you

closed 2024-02-06 by arjungupta72