ctu-mrs/mrs_uav_system #180: 🐞 Installtion error on Ubunto 20.04 ROS Noetic

MF-Ahmed, 2024-02-20:
Hi,
 After Adding the stable PPA curl https://ctu-mrs.github.io/ppa-stable/add_ppa.sh | bash
I get the following error after sudo apt install ros-noetic-mrs-uav-system-full

 sudo apt install ros-noetic-mrs-uav-system-full                                                                                                                    
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Thanks

klaxalk, 2024-02-20:
> The following information may help to resolve the situation:

missing "The following information that may help to resolve the situation" :-)

MF-Ahmed, 2024-02-20:
Here is the full message after curl https://ctu-mrs.github.io/ppa-stable/add_ppa.sh | bash
"λ › sudo apt install ros-noetic-mrs-uav-system-full                                                                                            ~
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-mrs-uav-system-full : Depends: python-is-python3 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
"
Thanks

klaxalk, 2024-02-20:
It looks like you are missing some important dependencies (-> have you installed ROS ??), or your package manager is in a broken state. The broken state can be sometimes easily fixed with `aptitude`, e.g., by trying
```bash
sudo aptitude install python-is-python3
```

MF-Ahmed, 2024-02-20:
Thanks, I did "sudo apt install python-is-python3"  and "sudo apt install ros-noetic-mrs-uav-system-full"
The following NEW packages will be installed:
  binutils-dev libio-pty-perl libipc-run-perl libncurses5 libqt5svg5-dev libqt5websockets5-dev libtime-duration-perl libtinfo5 libx264-dev
  moreutils mrs-uav-shell-additions ros-noetic-aloam-slam ros-noetic-catch-ros ros-noetic-ddynamic-reconfigure-python
  ros-noetic-dynamic-edt-3d ros-noetic-mav-msgs ros-noetic-mavlink-sitl-gazebo ros-noetic-mavros-extras ros-noetic-mrs-aloam-core
  ros-noetic-mrs-aloam-estimator-plugin ros-noetic-mrs-bumper ros-noetic-mrs-euler-counter-example ros-noetic-mrs-gazebo-common-resources
  ros-noetic-mrs-hector-core ros-noetic-mrs-hector-estimator-plugin ros-noetic-mrs-lib ros-noetic-mrs-modules-msgs ros-noetic-mrs-msgs
  ros-noetic-mrs-multireconfigure ros-noetic-mrs-multirotor-simulator ros-noetic-mrs-octomap-mapping-planning
  ros-noetic-mrs-octomap-planner ros-noetic-mrs-octomap-server ros-noetic-mrs-octomap-tools ros-noetic-mrs-open-vins-core
  ros-noetic-mrs-open-vins-estimator-plugin ros-noetic-mrs-pcl-tools ros-noetic-mrs-rviz-plugins ros-noetic-mrs-serial
  ros-noetic-mrs-subt-planning-lib ros-noetic-mrs-tf-connector ros-noetic-mrs-tf-estimator ros-noetic-mrs-tf-reconfigure
  ros-noetic-mrs-uav-autostart ros-noetic-mrs-uav-controllers ros-noetic-mrs-uav-coppelia-api ros-noetic-mrs-uav-coppelia-simulation
  ros-noetic-mrs-uav-core ros-noetic-mrs-uav-deployment ros-noetic-mrs-uav-dji-tello-api ros-noetic-mrs-uav-gazebo-simulation
  ros-noetic-mrs-uav-gazebo-testing ros-noetic-mrs-uav-hw-api ros-noetic-mrs-uav-managers ros-noetic-mrs-uav-modules
  ros-noetic-mrs-uav-path-loader ros-noetic-mrs-uav-px4-api ros-noetic-mrs-uav-state-estimators ros-noetic-mrs-uav-status
  ros-noetic-mrs-uav-system-full ros-noetic-mrs-uav-testing ros-noetic-mrs-uav-trackers ros-noetic-mrs-uav-trajectory-generation
  ros-noetic-mrs-uav-trajectory-loader ros-noetic-mrs-utils ros-noetic-mrs-vins-imu-filter ros-noetic-mrs-vins-republisher
  ros-noetic-nimbro-cam-transport ros-noetic-nimbro-log-transport ros-noetic-nimbro-net-monitor ros-noetic-nimbro-network
  ros-noetic-nimbro-service-transport ros-noetic-nimbro-tf-throttle ros-noetic-nimbro-topic-transport ros-noetic-nlopt
  ros-noetic-ouster-ros ros-noetic-ov-core ros-noetic-ov-eval ros-noetic-ov-init ros-noetic-ov-msckf ros-noetic-plotjuggler ros-noetic-px4
  ros-noetic-rviz-satellite ruby-erubis ruby-thor ruby-xdg tmuxinator
The following packages will be upgraded:
  libignition-math6 libignition-math6-dev ros-noetic-rosbag ros-noetic-tf2 ros-noetic-tf2-ros
5 upgraded, 87 newly installed, 0 to remove and 860 not upgraded.
Need to get 162 MB of archives.
After this operation, 263 MB of additional disk space will be used.
W: Sources disagree on hashes for supposely identical version '1.0.14' of 'mrs-uav-shell-additions:amd64'.
Do you want to continue? [Y/n] y
Get:1 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 libqt5svg5-dev amd64 5.12.8-0ubuntu1 [10.3 kB]
Get:2 http://fr.archive.ubuntu.com/ubuntu focal/main amd64 libio-pty-perl amd64 1:1.12-1 [32.4 kB]                            
Get:3 http://fr.archive.ubuntu.com/ubuntu focal/main amd64 libipc-run-perl all 20180523.0-2 [89.7 kB]
Get:4 http://fr.archive.ubuntu.com/ubuntu focal-updates/universe amd64 libtinfo5 amd64 6.2-0ubuntu2.1 [83.4 kB]
Get:5 http://fr.archive.ubuntu.com/ubuntu focal-updates/universe amd64 libncurses5 amd64 6.2-0ubuntu2.1 [96.9 kB]
Get:6 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 libqt5websockets5-dev amd64 5.12.8-0ubuntu1 [10.3 kB]
Get:7 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 libtime-duration-perl all 1.21-1 [13.1 kB]
Get:8 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 libx264-dev amd64 2:0.155.2917+git0a84d98-2 [569 kB]
Get:9 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 moreutils amd64 0.63-1 [60.5 kB]
Get:10 http://fr.archive.ubuntu.com/ubuntu focal-updates/main amd64 binutils-dev amd64 2.34-6ubuntu1.8 [3,634 kB]
Get:11 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-mrs-msgs 2:1.0.4-0focal.20240129.110426~git.7fbb469 [322 kB]
Get:12 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 ruby-erubis all 2.7.0-3 [88.6 kB]                 
Get:13 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 ruby-thor all 0.19.4-1 [43.9 kB]                                   
Get:14 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 ruby-xdg all 2.2.3-1 [10.2 kB]                                     
Get:15 http://fr.archive.ubuntu.com/ubuntu focal/universe amd64 tmuxinator all 1.1.4-3 [17.5 kB]                                 
Get:16 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-catch-ros amd64 0.5.0-2focal.20230620.184147 [427 kB]        
Get:17 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-tf2 2:0.7.666-0focal.20230917.065007~git.b496c7d [107 kB]
Get:18 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-mrs-lib 2:1.0.4-0focal.20240217.041620~git.5df5318 [345 kB]
Get:19 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-mrs-modules-msgs 2:1.0.0-0focal.20240120.041228~git.9b4f67e [120 kB]
Get:20 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-tf2-ros 2:0.7.666-0focal.20230917.065252~git.b496c7d [239 kB]
Get:21 https://ctu-mrs.github.io/ppa-unstable ./ ros-noetic-ouster-ros 2:0.5.2-0focal.20230917.091947~git.d0ce2c3 [278 kB]
[... cut ...]
E: Failed to fetch https://ctu-mrs.github.io/ppa-unstable/./ros-noetic-mrs-uav-gazebo-testing_1.0.0-0focal.20240220.044614~git.43ae3ff_amd64.deb  404  Not Found [IP: 185.199.110.153 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
λ › roscd mrs_uav_gazebo_simulation/tmux/one_drone                                                                                         ~
roscd: No such package 'mrs_uav_gazebo_simulation/tmux/one_drone'

even after sourcing the noetic workspace I'm not able to roscd into mrs_uav_gazebo_simulation

klaxalk, 2024-02-20:
Hey, to resolve the `404 Not Found [IP: 185.199.110.153 443]` error, do `sudo update update`.

closed 2024-02-20 by MF-Ahmed

klaxalk, 2024-02-21:
You can run ORB-SLAM2, that is for sure. I recommend starting with one of our drones presets from the `mrs_uav_gazebo_simulation`. Learn how to spawn them with different sensors (https://ctu-mrs.github.io/docs/simulation/gazebo/gazebo/drone_spawner.html) and find the ones that suit you. If you insist on a custom drone with custom sensor placement, follow the tutorial on how to create your custom drone package: https://ctu-mrs.github.io/docs/simulation/gazebo/gazebo/custom_drone.html.

The MRS system currently contains only one mapping & planning pipeline, the https://github.com/ctu-mrs/mrs_octomap_mapping_planning. However, you can connect other planning pipelines to the system using whatever output comes from the planning pipeline. The system accepts `path`, `trajectory`, or a `single point reference` as an input, so pretty much anything can be done (supposing the reference you generate is transformable by the TF library).

SLAM is only a technique to obtain localization. We integrate and use 4 different SLAMs/Odometries (ALOAM, LIOSAM, Hector, and OpenVins). You could possibly substitute the OpenVins with ORBSLam if you need to. However, to do frontier exploration, you need some volumetric mapping framework (like our [Octomap](https://github.com/ctu-mrs/mrs_octomap_mapping_planning)).

MF-Ahmed, 2024-02-21:
Thank you very much