ctu-mrs/mrs_uav_system #187: Installation issue with add_ppa.sh

MJavadZallaghi, 2024-08-05:
I want to install the MRS UVA system. When I run:
```bash
curl https://ctu-mrs.github.io/ppa-stable/add_ppa.sh | bash
```

I'll retrieve this issue:
```bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   952  100   952    0     0    297      0  0:00:03  0:00:03 --:--:--   297
bash: Adding MRS Stable PPA repository
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version (7.68.0-1ubuntu2.22).
dpkg-dev is already the newest version (1.19.7ubuntu3.2).
gpg is already the newest version (2.2.19-3ubuntu2.2).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
bash: "sudo curl -s --compressed -o /etc/apt/preferences.d/ctu-mrs-stable-preferences "https://ctu-mrs.github.io/ppa-stable/ctu-mrs-ppa-preferences.txt"" command failed with exit code 35
```

Is there any issue on the installation resources?

I'm on ROS Noetic.

Thanks in advance for any help.

cturmrsbot, 2024-08-29:
Hey, error code 35 means "A TLS/SSL connect error. The SSL handshake failed. The SSL handshake can fail due to numerous different reasons so the error message may offer some additional clues. Maybe the parties could not agree to a SSL/TLS version, an agreeable cipher suite or similar." Have you tried opening the file manually? Anyway, the server could have a hiccup. Try it again please...

cturmrsbot, 2024-08-29:
Alternatively, your system's time might have been wrong.

closed 2024-11-10 by klaxalk