ctu-mrs/mrs_uav_system #193: 🚀 Sending time/velocity with the trajectory

lidiaxp, 2024-12-02:
I’m looking to implement a system that allows a UAV to follow a predefined trajectory, including both position and time/velocity. I’ve found two possible approaches within the system:

1. Using the topic `/control_manager/trajectory_reference`, which accepts a message with the following structure:
```
trajectory:
  header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  input_id: 0
  use_heading: false
  fly_now: false
  loop: false
  dt: 0.0
  points:
  - position: {x: 0.0, y: 0.0, z: 0.0}
    heading: 0.0
```

2. Using the topic `/trajectory_generation/path`, which accepts a message with the following structure:
```
path:
  header:
    seq: 0
    stamp: {secs: 0, nsecs: 0}
    frame_id: ''
  input_id: 0
  use_heading: false
  fly_now: false
  stop_at_waypoints: false
  loop: false
  max_execution_time: 0.0
  max_deviation_from_path: 0.0
  dont_prepend_current_state: false
  override_constraints: false
  override_max_velocity_horizontal: 0.0
  override_max_acceleration_horizontal: 0.0
  override_max_jerk_horizontal: 0.0
  override_max_velocity_vertical: 0.0
  override_max_acceleration_vertical: 0.0
  override_max_jerk_vertical: 0.0
  relax_heading: false
  points:
  - position: {x: 0.0, y: 0.0, z: 0.0}
    heading: 0.0
```

In both cases, I’ve only been able to send positional data. How can I include velocity or timing information for each waypoint in the trajectory? Any advice or suggestions would be greatly appreciated!

kratkvit, 2024-12-03:
Hi Lidia, 

The topic _trajectory_generation/path_ is used for providing an input to the trajectory generation module responsible for generating feasible trajectory passing through the sequence of waypoints provided in the message while considering the provided or currently applied kinematic constraints. The first topic you have mentioned is used as an input to the trajectory tracking module and hence contains also the information about the timing of individual transition points in the form of trajectory sampling time (element _dt_ in the message _mrs_msgs/TrajectoryReference_). While this message allows you to specify the trajectory in the form of a sequence of positions and required timing, it does not allow the _dt_ to vary between individual transition points of the trajectory. Note that if you do not specify the _dt_, it is set to its default value, which is 0.2 s.

If you would need to specify the trajectory with varying sampling time between individual transition points, an alternative solution is to implement your own tracker module, which can provide a full state reference of type _mrs_msgs/TrackerCommand_ with varying time steps directly to controllers. However, note that this approach requires providing the reference to the controller at significantly higher rates than is common for trajectory generation approaches. It also replaces the MPC tracker, which is part of our system responsible for providing feasible trajectory reference to controllers.

lidiaxp, 2024-12-03:
In the `/control_manager/trajectory_reference` topic, will the _dt_ remain the same across all waypoints? I could use it to send my trajectory with time. My algorithm typically generates waypoints at 0.02-second intervals, so I believe this topic would work well for sending the trajectory.

Additionally, out of curiosity, would it be possible to modify the MPC tracker to accept the drone's full state? Or would it be simpler to create a new tracker altogether?

Thank you very much for your assistance.

kratkvit, 2024-12-03:
Yes, the _dt_ is constant for the whole horizon. The only constraint is that _dt_ has to be greater or equal to the tracker's internal step size, which is set to 0.01 s. 

Modifying the tracker should be possible, but it would result in changes in the optimization task and thus necessary updates of the solver. Depending on what you mean by full state, I expect smaller or bigger problems with real-time performance due to increasing dimensionality of the problem since the tracker is required to run at  ~100 Hz. Implementation-wise, most of the parts of the tracker would remain unchanged, so it would make sense to base the implementation of a new tracker on an existing MPC tracker rather than start the implementation from scratch.

lidiaxp, 2024-12-03:
I understand. Thank you very much for your assistance!

closed 2024-12-03 by lidiaxp